Z-Bolt/OctoScreen #201: Fix/Enhancement of How-To on Installing OctoScreen with a 3.5" 480x320 TFT screen

lmaertin, 2021-01-08:
Thanks for this great project and the detailed instructions for using OctoScreen with cheap 480x320 displays. I would like to give you some feedback on issues and possible enhancements on the instructions.

Issues:
- Minor: deb package url is pointing to an outdated version of octoscreen
- Major: By default the calibration of the touchscreen input (/usr/share/X11/xorg.conf.d/99-calibration.conf) swaps the axis. In both orientations of LCD-show (0° and 180°) this is not working. By seting the parameter to '0' all is working as expected. With swaped axis even a re-calibration of the touchscreen with xinput_calibrator leads to non-sense values...

Possible enhancements:
- During research, I found a tutorial to ease the whole handling with those displays:
https://www.raspberrypi.org/forums/viewtopic.php?t=143581
I'm not aware of possible pitfalls, but this could simplify the installation to some large extent. Maybe this is only working for larger screens (not tested yet!).
- Furthermore you may add a note that there are many variants of the 3.5" display which are supported by LCD-show. There are variants A, B (V1, v2), C and D. It's very easy to identify the display by the URLs to spec. pages on LCD-show's README.MD.

lmaertin, 2021-01-08:
Update, the enhancement from https://www.raspberrypi.org/forums/viewtopic.php?t=143581 is not working here, the LCD35 + fbturbo is needed to get it up and running.

eblieb, 2021-03-18:
/etc/X11/xorg.conf.d doesn't exist on my raspberry pi running OctoPi and OctoScreen. I got it working using the LCD35 + Fbturbo but I cant get the touch screen input to work properly.

felipebraz, 2022-05-09:
> /etc/X11/xorg.conf.d doesn't exist on my raspberry pi running OctoPi and OctoScreen. I got it working using the LCD35 + Fbturbo but I cant get the touch screen input to work properly.

Did you fixed it?