Z-Bolt/OctoScreen #206: Wiki enhancement for 3.5" TFT raspi cam issues

erdemozor, 2021-01-19:
I've followed the  [wiki](https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen) for setting up my tft display and noticed my raspi cam had stopped working for some reason.
After some trial and error I found that re enabling the raspberry camera module from the raspi-config tool fixed the issue.
Just thought this could be added to the wiki for others that could encounter this as it was a little frustrating.
Also this seems to be happening related to the drivers of the screen and I found a reoccurrance on the web using TochUI on a similar TFT.

thebeline, 2021-03-13:
Make and model of the TFT?  I did not run into this issue, but I am interested to learn more.

erdemozor, 2021-03-13:
> Make and model of the TFT? I did not run into this issue, but I am interested to learn more.

It was the Waveshare 3.5inch RPi LCD (either type A or B)

thebeline, 2021-03-13:
Hmmmm. I will try to reproduce.

JeffB42, 2021-03-13:
> Hmmmm. I will try to reproduce.

Thanks @thebeline !