Z-Bolt/OctoScreen #207: Moves are not working

theDrsh, 2021-01-20:
I just set up octoscreen with a hyperpixel(non-touch, which was a mistake on my part). I'm trying to run manual moves from the UI, I'm using a keyboard to do the input, other commands work fine, but for some reason the moves don't seem to work. I looked at the g-code terminal on octoprint and it shows no change, when I click the move buttons from within octoprint I see the gcode sent through that terminal. Is this something wrong with my setup?

JeffB42, 2021-02-01:
@theDrsh 

> hyperpixel(non-touch, which was a mistake on my part)

Ouch.  Yea, I wish they only sold the touch screen version. I think they also sell a 4" square version as well, which also adds confusion.  Maybe you can contact them and ask if you can exchange it for the touch version.

> for some reason the moves don't seem to work

Try homing the axis first, then moving.  That might be the issue - often times the printer won't move because it hasn't been homed yet (I think this is at the firmware and driver  level).

OctoScreen uses OctoPrint's REST APIs to get & set information, and for moving the printhead, it uses /api/printer/printhead (see https://docs.octoprint.org/en/master/api/printer.html#post--api-printer-printhead).  The code in question is 
`https://github.com/Z-Bolt/OctoScreen/blob/master/uiWidgets/MoveButton.go`
and `PrintHeadJogRequest` in 
`https://github.com/Z-Bolt/OctoScreen/blob/master/vendor/github.com/mcuadros/go-octoprint/printer.go`

Hope this helps,

Jeff

JeffB42, 2021-02-21:
@theDrsh is this still an issue?

theDrsh, 2021-03-02:
I upgraded to a Touch version and it seems to work, so I am not sure.

closed 2021-03-03 by JeffB42