Z-Bolt/OctoScreen #210: Error wihle installing octoscreen 2.6.1

Berblinger, 2021-01-26:
Hi,

i have a problem installing octoscreen. Missing dependencies are reported even though the requested packages are installed:

------------------------------------------------------------------------------
OctoPrint version : 1.5.3
OctoPi version    : 0.18.0
------------------------------------------------------------------------------
pi@octopi:~ $ sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils
[sudo] password for pi: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
xinit is already the newest version (1.4.0-1).
xserver-xorg is already the newest version (1:7.7+19).
libgtk-3-0 is already the newest version (3.24.5-1+rpt2).
x11-xserver-utils is already the newest version (7.7+8+rpi1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@octopi:~ $ sudo dpkg -i octoscreen_2.6.1_armhf.deb
Selecting previously unselected package octoscreen:armhf.
(Reading database ... 92330 files and directories currently installed.)
Preparing to unpack octoscreen_2.6.1_armhf.deb ...
Unpacking octoscreen:armhf (2.6.1-1) ...
dpkg: dependency problems prevent configuration of octoscreen:armhf:
 octoscreen:armhf depends on libatk1.0-0 (>= 1.12.4).
 octoscreen:armhf depends on libcairo-gobject2 (>= 1.10.0).
 octoscreen:armhf depends on libcairo2 (>= 1.14.0).
 octoscreen:armhf depends on libgdk-pixbuf2.0-0 (>= 2.25.2).
 octoscreen:armhf depends on libglib2.0-0 (>= 2.39.4).
 octoscreen:armhf depends on libgtk-3-0 (>= 3.21.5).
 octoscreen:armhf depends on libpango-1.0-0 (>= 1.29.4).
 octoscreen:armhf depends on libpangocairo-1.0-0 (>= 1.22.0).
 octoscreen:armhf depends on xserver-xorg.
 octoscreen:armhf depends on xinit.

dpkg: error processing package octoscreen:armhf (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 octoscreen:armhf

When I try to check the individual packages, the following message appears:
pi@octopi:~ $ sudo apt-get install libatk1.0-0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libatk1.0-0 is already the newest version (2.30.0-2).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 octoscreen:armhf : Depends: libatk1.0-0:armhf (>= 1.12.4) but it is not going to be installed
                    Depends: libcairo-gobject2:armhf (>= 1.10.0) but it is not going to be installed
                    Depends: libcairo2:armhf (>= 1.14.0) but it is not going to be installed
                    Depends: libgdk-pixbuf2.0-0:armhf (>= 2.25.2) but it is not going to be installed
                    Depends: libglib2.0-0:armhf (>= 2.39.4) but it is not going to be installed
                    Depends: libgtk-3-0:armhf (>= 3.21.5) but it is not going to be installed
                    Depends: libpango-1.0-0:armhf (>= 1.29.4) but it is not going to be installed
                    Depends: libpangocairo-1.0-0:armhf (>= 1.22.0) but it is not going to be installed
                    Depends: xserver-xorg:armhf but it is not going to be installed
                    Depends: xinit:armhf but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

When I check the missing packages after uninstalling octoscreen, the packages are reported as installed:
pi@octopi:~ $ sudo dpkg --purge octoscreen
(Reading database ... 92442 files and directories currently installed.)
Removing octoscreen:armhf (2.6.1-1) ...
Purging configuration files for octoscreen:armhf (2.6.1-1) ...
pi@octopi:~ $ sudo apt-get install libatk1.0-0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libatk1.0-0 is already the newest version (2.30.0-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@octopi:~ $

Hofei90, 2021-01-29:
Do you try this?

`E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).`

Berblinger, 2021-01-29:
> Do you try this?
> 
> `E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).`

Yes, tried that, with octoscreen installed it removes nearly all of the packages octoscreen depends on, without octoscreen installed it reports that everything is just fine.

It seams that octoscreen breaks the package management.

JeffB42, 2021-01-31:
@Berblinger I don't know what to tell you, as it's worked for me, and I've installed many times.  My only suggestions are:
1) start with a clean image.  Download the latest OctoPi, re-image the SD card with OctoPi
2) Follow the steps at https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

lupos, 2021-02-01:
is this a pi 4b? I am having the same issue and can't get anything to work. also i cant use the latest release of octopi (as of a few days ago) but the latest nightly works for me to get octoprint running.

thebeline, 2021-03-13:
@Berblinger || @lupos - Have you had any luck resolving this issue?  

I do seem to recall running into a similar error on one of my systems once, but have not since, and for the life of me I can't seem to recreate the issue.

If I recall, I THINK I resolved it by `sudo apt --fix-broken install` while OS was installed (yes, removes everything) then I ran the install again?  Or maybe it was by following the LCD instructions, and first running:

```
$ sudo apt-get update
$ sudo apt-get install -f
```

And then the OS stuff...  I think that was the fix, actually.

closed 2021-04-10 by JeffB42