Z-Bolt/OctoScreen #211: Octoscreen not being displayed

uli6, 2021-01-27:
Hello, amazing people!

I recently bought a 3.5-inch screen for my octoprint running on a Rpi 4. ([this one](shorturl.at/xABM2)). The driver installation was ok, and it started showing information (terminal) on the screen. At this point in time, I was really confident.

So,  I installed OctoScreen and nothing happens, so, I reinstalled octoprint and then installed Octoscreen again, and nothing.

I've tried a lot of stuff I found on the internet and nothing works.

The service is running, octoprint is running (and printing), but the application is not being displayed.

[image]

[image]

markeby, 2021-02-13:
Confirm that you have installed all of the components for X11 and GTK-3.  It appears that the X11 client isn't running.

JeffB42, 2021-02-16:
@uli6 I agree with @markeby prognosis - I think X11 and GTK-3 either isn't installed or isn't installed correctly.

thebeline, 2021-03-13:
@uli6 - Have you had success?  Is this resolved?

JeffB42, 2021-04-10:
Closed due to last of response.

closed 2021-04-10 by JeffB42