Z-Bolt/OctoScreen #213: Buzzer every touch

Meinsda, 2021-01-29:
is it possible to add a buzzer everytime i touch the touch screen so that i have a response from octoprint that i accept my input?

JeffB42, 2021-01-31:
Well, the RPi doesn't have a buzzer built into it.  I suppose I might be able to play a sound file and have the sound play out the audio jacks.  I don't know... I'll add it to the feature request list though.

Meinsda, 2021-02-01:
Nice ;) But to add a buzzer to a raspberry is not a problem... Add only a buzzer to an gpio pin like this one https://bit.ly/39yYPCM . But you must modulate the signal for the output!

Dweinbach, 2024-04-12:
rather than a buzzer, why not use the built-in  audio of the pi? there is an audio output jack, but my ElecLab " screen has built in speakers.
I don't know anything about the pi os audio support, but you can test with with this native command:
speaker-test -c2 -twav -l7

I agree that adding audible feedback would be a big improvement to octoscreen.

Ant pro-tip on where a line could be added to play a sound whenever a on-screen "key" is pressed?

Dweinbach, 2024-04-12:
after a little digging you can install "sox" and use the "play" command to simply play a sound as well.