Z-Bolt/OctoScreen #214: Temporary disable OctoScreen

ingo916, 2021-02-01:
I am having issues with a plugin that is causing the HTTP service in OctoPrint to stop. The problem goes away when I am in OctoPrint SAFE MODE. I am going to disable my plugins one by one to try and identify which one is causing the issue.

Is there a way to disable the OctoScreen plugin from loading without having to uninstall it?

JeffB42, 2021-02-04:
I'm going off of memory here...  SSH into your RPi, then:
`$ nano /etc/rc.local`
or
`$ nano /lib/systemd/system/octoscreen.service`
...and you should be able to comment out the service and prevent it from auto-running.  Then, run `$ sudo systemctl restart octoscreen.service` to re-run it.

I'm not 100% sure about this... I'm going off of memory here, but this should get you pointed in the right direction.

markeby, 2021-02-13:
You are correct @JeffB42.  Just commend out the [Service] portion of the file and reboot.

JeffB42, 2021-02-16:
@ingo916 is this still an issue?

ingo916, 2021-02-16:
No please close. Thanks.

Ingo Azarvand
209-207-2243
Licensed FFL Dealer

closed 2021-02-16 by JeffB42