Z-Bolt/OctoScreen #215: OctoScreen (I think) is using up all the threads for my octoprint

dkowis, 2021-02-02:
Since adding octoscreen to my setup, every time I come back to print stuff on my printer, I have to restart octoprint from the command line. 

My operating mode is to leave the octopi on all the time, and just connect it when I'm ready to fire up prints. The pi doesn't power off. Unfortunately, since adding octoscreen, I seem to have run out of threads in the app. Sometimes I can't even issue a reboot command over the webui. 

I was filing this here, because I seem to be the only one with the problem, and I'm wondering if I'm rare in how I run the thing, or if I have a different problem.

acogburn, 2021-02-02:
I had to give up on octoscreen.. Octodash is my new go to. Have yet to find
a bug

JeffB42, 2021-02-04:
@dkowis At the moment, there's a known bug where, on some systems, if OctoPrint and OctoScreen are running but the printer is turned off or disconnected, the system restarts.  I'll probably fix this in 2.8, but for the mean time, the workaround is to leave your printer on a be idle.

JeffB42, 2021-02-16:
@dkowis I attempted to reproduce this bug, but I'm not able to.  I know there's something there, i just haven't been able to reproduce it.

* Which version of OctoScreen are you running?  If not 2.6.1, please update to 2.6.1.
* Which model of Raspberry Pi are you using?
* Have you turned logging on in OctoScreen?
* Clear out OctoScreen's log file, and then with OctoPrint and OctoScreen on, but with your printer off, are you seeing any errors in OctoScreen's log file?

dkowis, 2021-02-16:
```
OctoPrint version : 1.5.3
OctoPi version    : 0.17.0

pi@octopi:~ $ dpkg -l octoscreen
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-================================================================
ii  octoscreen     2.6.1-1      armhf        A OctoPrint touch interface for TFT touch modules based on GTK+3
```

It is an RPi4B 1Gig:  `Raspberry Pi 4 Model B Rev 1.1` I think I have the 1 gig model for this:

```
pi@octopi:~/logs $ free -m
              total        used        free      shared  buff/cache   available
Mem:            803         136         430           9         237         602
Swap:            99           0          99
```

right after startup, only a few minutes, haven't interacted with the UI or anything. Printer is off, octopi is up. I think it is restarting occasionally. 

```
time="2021-02-16T04:36:09Z" level=debug msg="OctoScreen - entering main.init()" file=main.go func=main.init.0 line=55
time="2021-02-16T04:36:09Z" level=debug file=main.go func=main.readConfig line=169
time="2021-02-16T04:36:09Z" level=debug file=main.go func=main.readConfig line=170
time="2021-02-16T04:36:09Z" level=debug msg="entering main.readConfig()" file=main.go func=main.readConfig line=171
time="2021-02-16T04:36:09Z" level=info msg="Path to OctoPrint's config file: \"/home/pi/.octoprint/config.yaml\"" file=main.go func=main.readConfig line=180
time="2021-02-16T04:36:09Z" level=info msg="main.readConfig() - ReadFile() succeeded" file=main.go func=main.readConfig line=187
time="2021-02-16T04:36:09Z" level=info msg="main.readConfig() - YAML config file was decoded" file=main.go func=main.readConfig line=193
time="2021-02-16T04:36:09Z" level=info msg="main.readConfig() - server host is: \"localhost\"" file=main.go func=main.readConfig line=200
time="2021-02-16T04:36:09Z" level=info msg="main.readConfig() - server port is: 5000" file=main.go func=main.readConfig line=206
time="2021-02-16T04:36:09Z" level=debug msg="leaving main.readConfig()" file=main.go func=main.readConfig line=209
time="2021-02-16T04:36:09Z" level=debug file=main.go func=main.readConfig line=210
time="2021-02-16T04:36:09Z" level=debug file=main.go func=main.readConfig line=211
time="2021-02-16T04:36:09Z" level=debug msg="OctoScreen - entering main.setApiKey()" file=main.go func=main.setApiKey line=128
time="2021-02-16T04:36:09Z" level=debug msg="main.setApiKey() - APIKey is empty, now using cfg.API.Key" file=main.go func=main.setApiKey line=132
time="2021-02-16T04:36:09Z" level=debug msg="main.setApiKey() - APIKey is \"B46---C33\"" file=main.go func=main.setApiKey line=141
time="2021-02-16T04:36:09Z" level=debug msg="OctoScreen - leaving main.setApiKey()" file=main.go func=main.setApiKey line=144
time="2021-02-16T04:37:52Z" level=debug msg="OctoScreen - entering main.init()" file=main.go func=main.init.0 line=55
time="2021-02-16T04:37:52Z" level=debug file=main.go func=main.readConfig line=169
time="2021-02-16T04:37:52Z" level=debug file=main.go func=main.readConfig line=170
time="2021-02-16T04:37:52Z" level=debug msg="entering main.readConfig()" file=main.go func=main.readConfig line=171
time="2021-02-16T04:37:52Z" level=info msg="Path to OctoPrint's config file: \"/home/pi/.octoprint/config.yaml\"" file=main.go func=main.readConfig line=180
time="2021-02-16T04:37:52Z" level=info msg="main.readConfig() - ReadFile() succeeded" file=main.go func=main.readConfig line=187
time="2021-02-16T04:37:52Z" level=info msg="main.readConfig() - YAML config file was decoded" file=main.go func=main.readConfig line=193
time="2021-02-16T04:37:52Z" level=info msg="main.readConfig() - server host is: \"localhost\"" file=main.go func=main.readConfig line=200
time="2021-02-16T04:37:52Z" level=info msg="main.readConfig() - server port is: 5000" file=main.go func=main.readConfig line=206
time="2021-02-16T04:37:52Z" level=debug msg="leaving main.readConfig()" file=main.go func=main.readConfig line=209
time="2021-02-16T04:37:52Z" level=debug file=main.go func=main.readConfig line=210
time="2021-02-16T04:37:52Z" level=debug file=main.go func=main.readConfig line=211
time="2021-02-16T04:37:52Z" level=debug msg="OctoScreen - entering main.setApiKey()" file=main.go func=main.setApiKey line=128
time="2021-02-16T04:37:52Z" level=debug msg="main.setApiKey() - APIKey is empty, now using cfg.API.Key" file=main.go func=main.setApiKey line=132
time="2021-02-16T04:37:52Z" level=debug msg="main.setApiKey() - APIKey is \"B46---C33\"" file=main.go func=main.setApiKey line=141
time="2021-02-16T04:37:52Z" level=debug msg="OctoScreen - leaving main.setApiKey()" file=main.go func=main.setApiKey line=144
```

Yeah that definitely repeats, so it's .... restarting?
```
pi@octopi:~ $ systemctl status octoscreen
● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; disabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-02-16 04:39:46 GMT; 10s ago
  Process: 7092 ExecStartPost=/bin/bash /etc/octoscreen/disablescreenblank.sh 0 (code=exited, status=0/SUCCESS)
 Main PID: 7078 (xinit)
    Tasks: 11 (limit: 1309)
   CGroup: /system.slice/octoscreen.service
           ├─7078 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor
[... cut ...]
I see something in dmesg about `broken atomic modeset userspace detected, disabling atomic` I have [taken a couple mitigating steps](https://www.raspberrypi.org/forums/viewtopic.php?p=1742818&sid=b7fdf210f23cc75d3eacf72a0a65663d#p1742818) to see if I can make that go away.

Doing those steps didn't, it still complains about "broken atomic modeset userspace detected, disabling atomic"

I turned the logging level up to debug, and will attach it.
[logfile.txt](https://github.com/Z-Bolt/OctoScreen/files/5986161/logfile.txt)

Thanks!

JeffB42, 2021-02-16:
> if I don't respond for a few days, it's because everything is frozen.

LOL.  Yep, the country is getting hammered.

First, no problem, and thanks for the help!
Second, no worries about any delay... I volunteer on this and I've had to put in place some boundaries, and I (try to) only look at the issues and reply on weekends.

Some thoughts:
1) this is a bit disconcerting.  There aren't any errors in the log file, so it's a mystery to me at the moment.
2) The only thing that I see that's off, is that in the beginning there is:
```
time="2021-02-16T04:46:25Z" level=debug msg="main.setApiKey() - APIKey is \"B46---C33\"" file=main.go func=main.setApiKey line=141
```
...and then later there is:
```
time="2021-02-16T04:46:26Z" level=info msg="key: \"OCTOPRINT_APIKEY\", value: \">>M---G<<\"" file=environment.go func=utils.dumpObfuscatedEnvironmentVariable line=118
```
The value here is actually ">>MISSING<<",  but it's obfuscated, and it being ">>M---G<<" instead of ">>MISSING<<" isn't an issue, but the fact that it's missing, where as earlier it was found is odd.  It might be a bug, but I'm not sure this is causing the issue you are seeing.

Try this:
- SSH into your RPi
- Edit your OctoScreen config file, and set the OCTOPRINT_APIKEY value to your API key in the config file as well
- Reboot
...this shouldn't change anything, but who knows, and if it does fix the issue, it give me something to zero in on.

Another thing to try is this: try disabling the OctoScreen service (see https://github.com/Z-Bolt/OctoScreen/issues/214).  With OctoScreen disabled, and your printer off, boot your RPi+OctoPrint.  Does it still reboot over and over on its own?

Thanks again for the help,

Jeff

dkowis, 2021-02-16:
> Some thoughts:
>     1. this is a bit disconcerting.  There aren't any errors in the log file, so it's a mystery to me at the moment.
>     2. The only thing that I see that's off, is that in the [...]

Trying that...

It could've been that the OCTOPRINT_APIKEY variable is not commented out, just empty, and also the OCTOPRINT_CONFIG_FILE was also specified. I have entered the API key in the value, and will let it run for a while and then give you another log file.

I am seeing an eventual: 

```
time="2021-02-16T07:04:08Z" level=debug msg="leaving ui.update() - connectionAttempts > 8" file=ui.go func="ui.(*UI).update" line=368
```

and thats when the application seems to "restart" and it reloads everything. I didn't see anything inthere indicating the MISSING like you mentioned before.

Here's another log file:
[logfile.txt](https://github.com/Z-Bolt/OctoScreen/files/5986709/logfile.txt)

> ...this shouldn't change anything, but who knows, and if it does fix the issue, it give me something to zero in on.

Yeah it didn't appear to change anything :(

> 
> Another thing to try is this: try disabling the OctoScreen service (see #214). With OctoScreen disabled, and your printer off, boot your RPi+OctoPrint. Does it still reboot over and over on its own?
> 
> And one last thing: With your RPi, what's plugged into it? I've been assuming that the only thing connected is the display, and nothing is plugged into any of the USB ports or HDMI port. If anything else is connected, remove them and see if the issue still happens.

The pi doesn't reboot, sorry, didn't mean to convey that. The octoprint software keeps restarting. I never had any reboot issues of the Pi Itself. I currently have octoscreen disabled, and I start it manually to test this stuff, like you're describing.

JeffB42, 2021-02-20:
@dkowis I added some additional guards (and a lot of additional logging) and I think this issue is fix with the 2.7 release.  I'll probably release 2.7 in a few weeks so keep an eye out for it.

cghdev, 2021-02-25:
I've been having the same issue when I leave my printer off and octoprint on. The octoscreen service keeps restarting.
Taking a look to journalctl, it's been killed by the watchdog for some reason. I increased the watchdog's timeout to 1min and, even though it takes a bit longer to kill it, it ends up killing the process too.
```bash
Feb 25 16:05:08 octopi xinit[615]: time="2021-02-25T16:05:08+01:00" level=debug msg="entering ui.checkNotification()" file=ui.go func="ui.(*UI).checkNotification" line=265
Feb 25 16:05:08 octopi systemd[1]: Started OctoScreen.
Feb 25 16:05:08 octopi xinit[615]: time="2021-02-25T16:05:08+01:00" level=debug msg="leaving ui.checkNotification()" file=ui.go func="ui.(*UI).checkNotification" line=284
Feb 25 16:05:08 octopi xinit[615]: time="2021-02-25T16:05:08+01:00" level=debug msg="entering ui.verifyConnection()" file=ui.go func="ui.(*UI).verifyConnection" line=152
Feb 25 16:05:08 octopi xinit[615]: time="2021-02-25T16:05:08+01:00" level=debug msg="entering ui.sdNotify()" file=ui.go func="ui.(*UI).sdNotify" line=406
Feb 25 16:05:08 octopi xinit[615]: time="2021-02-25T16:05:08+01:00" level=debug msg="leaving ui.sdNotify()" file=ui.go func="ui.(*UI).sdNotify" line=415
Feb 25 16:05:14 octopi xinit[615]: time="2021-02-25T16:05:14+01:00" level=error msg="ui.verifyConnection() - s.Current.State is IsOffline, and (ConnectRequest)Do(UI.Client) returned an error: \"Post http://localhost:5000/api/connection: net/http: request canceled (Client.Timeout exceeded while awaiting headers)\"" file=logger.go func=utils.LogError line=86
Feb 25 16:05:14 octopi xinit[615]: time="2021-02-25T16:05:14+01:00" level=info msg="ui.verifyConnection() - newUIState equals ui.UIState and is: \"splash\"" file=ui.go func="ui.(*UI).verifyConnection" line=233
Feb 25 16:05:14 octopi xinit[615]: time="2021-02-25T16:05:14+01:00" level=debug msg="leaving ui.verifyConnection()" file=ui.go func="ui.(*UI).verifyConnection" line=234
Feb 25 16:05:14 octopi xinit[615]: time="2021-02-25T16:05:14+01:00" level=debug msg="leaving ui.update()" file=ui.go func="ui.(*UI).update" line=402
Feb 25 16:05:29 octopi xinit[615]: time="2021-02-25T16:05:29+01:00" level=debug msg="entering ui.update()" file=ui.go func="ui.(*UI).update" line=363
Feb 25 16:05:29 octopi xinit[615]: time="2021-02-25T16:05:29+01:00" level=info msg="ui.update() - thus.UIState is:  splash" file=ui.go func="ui.(*UI).update" line=372
Feb 25 16:05:29 octopi xinit[615]: time="2021-02-25T16:05:29+01:00" level=debug msg="entering ui.checkNotification()" file=ui.go func="ui.(*UI).checkNotification" line=265
Feb 25 16:05:29 octopi systemd[1]: octoscreen.service: Watchdog timeout (limit 10s)!
Feb 25 16:05:29 octopi xinit[615]: (EE)
Feb 25 16:05:29 octopi xinit[615]: (EE) Backtrace:
Feb 25 16:05:29 octopi xinit[615]: (EE)
Feb 25 16:05:29 octopi xinit[615]: (EE) Received signal 6 sent by process 1, uid 0
Feb 25 16:05:29 octopi xinit[615]: (EE)
Feb 25 16:05:29 octopi xinit[615]: Fatal server error:
Feb 25 16:05:29 octopi xinit[615]: (EE) Caught signal 6 (Aborted). Server aborting
Feb 25 16:05:29 octopi xinit[615]: (EE)
Feb 25 16:05:29 octopi xinit[615]: Please consult the The X.Org Foundation support
Feb 25 16:05:29 octopi xinit[615]:          at http://wiki.x.org
Feb 25 16:05:29 octopi xinit[615]:  for help.
Feb 25 16:05:29 octopi xinit[615]: (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
Feb 25 16:05:29 octopi xinit[615]: (EE)
Feb 25 16:05:29 octopi systemd[1]: octoscreen.service: Killing process 615 (xinit) with signal SIGABRT.
Feb 25 16:05:29 octopi xinit[615]: SIGABRT: abort
Feb 25 16:05:29 octopi xinit[615]: PC=0xaf478 m=0 sigcode=0
Feb 25 16:05:29 octopi xinit[615]: signal arrived during cgo execution
Feb 25 16:05:29 octopi xinit[615]: goroutine 1 [select, locked to thread]:
Feb 25 16:05:29 octopi xinit[615]: runtime.futex(0xca9148, 0x0, 0x0, 0x0, 0x0, 0x0, 0xca9148, 0x0, 0x0, 0x8a268, ...)
Feb 25 16:05:29 octopi xinit[615]:         /usr/local/go/src/runtime/sys_linux_arm.s:246 +0x1c
Feb 25 16:05:29 octopi xinit[615]: runtime.futexsleep(0xca9148, 0x0, 0xffffffff, 0xffffffff)
Feb 25 16:05:29 octopi xinit[615]:         /usr/local/go/src/runtime/os_linux.go:45 +0x74
Feb 25 16:05:29 octopi xinit[615]: runtime.notesleep(0xca9148)
Feb 25 16:05:29 octopi xinit[615]:         /usr/local/go/src/runtime/lock_futex.go:151 +0xac
Feb 25 16:05:29 octopi xinit[615]: runtime.stoplockedm()
Feb 25 16:05:29 octopi xinit[615]:         /usr/local/go/src/runtime/proc.go:1827 +0x64
Feb 25 16:05:29 octopi xinit[615]: runtime.schedule()
Feb 25 16:05:29 octopi xinit[615]:         /usr/local/go/src/runtime/proc.go:2213 +0x280
Feb 25 16:05:29 octopi xinit[615]: runtime.park_m(0x12a000e0)
Feb 25 16:05:29 octopi xinit[615]:         /usr/local/go/src/runtime/proc.go:2318 +0x90
Feb 25 16:05:29 octopi xinit[615]: runtime.mcall(0x7ebccb18)
Feb 25 16:05:29 octopi xinit[615]:         /usr/local/go/src/runtime/asm_arm.s:200 +0x5c
Feb 25 16:05:29 octopi xinit[615]: goroutine 19 [select]:
Feb 25 16:05:29 octopi xinit[615]: github.com/Z-Bolt/OctoScreen/utils.(*BackgroundTask).loop(0x12a739c0)
Feb 25 16:05:29 octopi xinit[615]:         /go/src/github.com/Z-Bolt/OctoScreen/obj-arm-linux-gnueabihf/src/github.com/Z-Bolt/OctoScreen/utils/background_task.go:60 +0x100
Feb 25 16:05:29 octopi xinit[615]: created by github.com/Z-Bolt/OctoScreen/utils.(*BackgroundTask).Start
[... cut ...]
Not sure what's making the watchdog kill the process. If I stop the service and run octoscreen manually like this
```bash
sudo OCTOPRINT_HOST=http://localhost:5000 OCTOSCREEN_STYLE_PATH=/opt/octoscreen/styles/z-bolt/ OCTOPRINT_CONFIG_FILE=/home/pi/.octoprint/config.yaml OCTOSCREEN_LOG_LEVEL=error OCTOSCREEN_RESOLUTION=800x480 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor
```
it seems to work fine.

This is on a Raspberry Pi 3 Model B Rev 1.2, running OctoPrint 1.5.3 and OctoSCreen 2.6.1

JeffB42, 2021-02-27:
I've looked into this and fixed the issue (I think I got most of them, anyway).  The fixes will be in the upcoming 2.7.0 release which should be out in a few weeks,  in the meantime, I suggest running with the printer on.  It doesn't have to be heated, it just has to be connected to OctoPrint.

cghdev, 2021-02-27:
Thank you @JeffB42!

JeffB42, 2021-03-27:
v2.7.0 was just release.  If this is still an issue, or if you find any others with the release, please post let me know.

closed 2021-03-27 by JeffB42