Z-Bolt/OctoScreen #216: [Feature Request] Preheat for Auto Level

hawkeyexp, 2021-02-02:
Hi, it would be great if octoscreen does an auto preheat of bed for leveling :-)

thebeline, 2021-02-10:
I'll second this.  However, I have the SmartPreheat Plugin, and I have the bed leveling handled there every print.

One downside to this feature (and what Smart Preheat handles) is that every filament may have different heat settings, and therefore adding the "preheat" to the standard level is not really a thing...  Much better to have it handled either in startup G-Code from your slicer, or the way I do with Smart Preheat.

hawkeyexp, 2021-02-11:
I agree we have different temps for different materials but currently autoleveling does not trigger any preheat - so it starts leveling with a cold bed (if you forget to preheat and you can't interrupt) - so we should possible get a dialog  with predefined values to select from and after this temp is reached the autolevel should start - this way we have it in one workflow :-)

thebeline, 2021-02-11:
Options:

- Add auto-level after pre-heat in Smart Preheat Plugin
- If **NOT** using OctoPrint and the Smart Preheat plugin:
  - Add auto-level into your start g-code in your slicer, after the pre-heat (and wait, if you are doing that).
  - Create some g-code files that have pre-heat and auto-level for predefined temps, and run those (no actual print, just a "blank" g-code, with heat-up, wait, and auto-level).

thebeline, 2021-02-11:
Also: See [Menu Configuration](https://github.com/Z-Bolt/OctoScreen#menu-configuration), where you can add custom buttons to do this.

For example, the g-code to execute to preheat to 245 extruder and 85 bed:
```
M190 R85 ;   Set Bed to 85C and wait
G4 S300 ; Wait 6 minutes (in seconds)
M109 R245 ; Set Extruder to 245C and wait
G28 ; Home all axes
G29 ; Auto bed-level (BL-Touch)
G92 E0 ; Reset Extruder
```
I think, I may have missed some stuff, but that's the gist.

thebeline, 2021-02-11:
Also See: https://www.reddit.com/r/ender3/comments/aiebc7/better_cura_start_gcode_for_bltouch/

hawkeyexp, 2021-02-11:
Hey @thebeline sorry but it has nothing to do with the cura start codes and i have no motivation to discuss about your misunderstanding - we have an option to manually start a leveling via octoscreen which has nothing to do with the startcode of cura and this should make sure that the bd is heated nothing more nothing less...

thebeline, 2021-02-11:
@hawkeyexp - I am not misunderstanding you, I understand quite well what you are requesting, and I am considering it from a programming perspective with that.  I am just suggesting that adding a pre-heat option to the auto-level, due to specialty considerations with regards to individual set-ups, presents UX issues, and the desired result can be achieved a number of different ways (by, as mentioned, either setting up an auto-level that triggers after pre-heat for any/all prints via a plugin or configured start-up g-code), or with specially tailored g-code files that do precisely an auto-level, that you just "print" at will, as opposed to complicating the UI to accommodate all unique needs.

That's all I am saying.

Example G-Code file:
[Auto Level 245x85.gcode.txt](https://github.com/Z-Bolt/OctoScreen/files/5968350/Auto.Level.245x85.gcode.txt)

Also, don't forget to update your slicer to LOAD the mesh:
```
M420 S1 ; Load the ABL mesh
; M420 S1 Z10 ; Load the ABL mesh and fade it out over 10mm in Z
```

markeby, 2021-02-13:
I do a parallel preheat and auto level on all of my prints as part of the default gcode in the slicer.

thebeline, 2021-02-15:
@hawkeyexp I think I found a true solution!

So, after some digging, I found that all of this and more is possible with the [OctoPrint Plugin Manager for OctoScreen](https://github.com/Z-Bolt/OctoPrint-Z-Bolt-OctoScreen).

First thing in there is the Auto Bed Leveling G-Code editor.

Now, for reference; prior to finding THIS, I had whipped up the following OctoPrint Custom Control to handle what you were asking, assuming it would auto-populate into OctoScreen (I was wrong), but let's use it for quick reference right now:

```
controls:
  - name: Automatic Bed Leveling
    layout: vertical
    children:
    - name: Pre-Heat and ABL
      confirm: You are about to begin Pre-Heat and Automatic Bed Leveling.
      commands:
      - M117 Set Bed Temp %(abl_preheat_temp)iC
      - M140 S%(abl_preheat_temp)i
      - G28
      - G1 Z100.0 F1500
      - M117 Waiting for Bed to settle at Temp %(abl_preheat_temp)iC
      - M190 S%(abl_preheat_temp)i
      - M117 Waiting %(abl_preheat_wait)i seconds for Bed to Settle
      - G4 S%(abl_preheat_wait)i
      - M117 Auto Bed-Level
      - G29
      - G1 Z2.0 F1500
      - M500
      - M117 Reseting system and turning off heat
      - G92 E0
      - M104 S0
      - M140 S0
      - M107
      - M117 Pre-Heat and Auto Bed-Level Complete!
      input:
      - name: Bed Temperature
        default: 85
        parameter: abl_preheat_temp
        slider:
          max: 100
          min: 0
          step: 5
      - name: Wait Time (in Seconds)
        default: 360
        parameter: abl_preheat_wait
        slider:
          max: 600
          min: 0
          step: 60
```

Taking the above, and looking at the ABL G-Code option in the OctoScreen manager, we could change what it currently says to something similar to what I had written above:
```
M117 Set Bed Temp 85C
M140 S85
G28
G1 Z100.0 F1500
M117 Waiting for Bed to settle at Temp 85C
M190 S85
M117 Waiting 6 minutes for Bed to Settle
G4 S360
M117 Auto Bed-Level
G29
G1 Z2.0 F1500
M500
M117 Reseting system and turning off heat
G92 E0
M104 S0
M140 S0
M107
M117 Pre-Heat and Auto Bed-Level Complete!
```

You have now custom configured the ABL with pre-heat!!

There are also ways to add custom controls and screens to OctoScreen via the manager as well, and I am looking in to those now.  Should be able to add a fully featured ABL with pre-heat select, there.  I'll let you know!

hawkeyexp, 2021-02-15:
Sounds nice...

thebeline, 2021-02-15:
> Sounds nice...

I can confirm it works as intended.  I set the above code in that field, and ran the auto-level from OctoScreen and the printer behaved as expected.  G2G I think...  

I am currently hashing out a weird parse issue with custom controls, but I should have that sorted shortly...

thebeline, 2021-02-15:
Yeah, NVM.  There is an issue in an upstream library that parses Custom Action Inputs incorrectly, and it also doesn't seem like the OctoScreen code really DOES anything with the Inputs, so...  Yeah.

I am not dealing with that right now, even though it seems super neat, I have already wasted too much time.  With regards to your Pre-Heat request, however, I think this about resolves it.