Z-Bolt/OctoScreen #218: OctoScreen disappears when a printer is plugged in

tre4b, 2021-02-07:
I have an RPI 4 4gb, and a 3.5inch waveshare compatible screen.

I followed your install instructions and built the pi ready, but not attached to my printer and all seemed okay.  There was a odd infrequent reloading of the graphics with a brief glimpse of the underlying shell.   And there was an icon on the info page that showed octo screen  but beside it another icon with Plugin not found.

I assumed all this was because I was not plugged into the computer.  With it plugged into the computer octoscreen has vanished and all I see is the shell.  Octoprint has correctly connected to the printer though and can control it.  The shell screen keeps going regularly black, asthough Octoscreen is attempting to start.  Then it goes back to the shell again.

tre4b, 2021-02-07:
Testing now without the printer plugged in and OctoScreen still refuses to display.  Have tried working my way through the install instructions again but no dice.  Cannot figure out where to find the logs. 

Config says logs/logfile.txt   and level = warn.   Not sure what other levels could be applied or where that file could be.  There is no log directory in /etc/OctoScreen,   nothing in .octoprint/logs  and nothing in /var/logs.

tre4b, 2021-02-08:
Tonight just tried the whole thing from scratch.   After all was done and I added the octoprint API Key I then rebooted.

I did all of this with the printer plugged in.  On that final reboot OctoScreen showed up and I thought it was working but then the printer finished initialising and the OctoScreen disappeared again leaving me only with the login prompt.  Looks like OctoScreen doesn't work for me :-(   This is all out of the box with the only octopi change being creating an API key.

tre4b, 2021-02-09:
More self reporting.  I think that this is something to do with Access Control.  This seems to be forced on on the Octopi build I am using
Octoprint = Version 1.5.3
OctoPi = Version 0.18.0,

The OCTOPRINT_CONFIG_FILE=  is set correctly for my local build but the key is not found.

If I do not supply an API key then I get "Required Environmental variable is not set"  OCTOPRINT_APIKey stuck on my screen.  This presumably is OctoScreen displaying this on a screen.

Octoprint tells me not to use the General API key and I should use Access Aplication keys instead.  If I create an OctoScreen Access API key and provide it then I get no OctoScreen,  just the pi linux login.   The console is overwritten briefly by a blank screen about once every 30 seconds.

tre4b, 2021-02-09:
Found this in the logs from in Octoprint.log

2021-02-09 21:31:14,356 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 5.09ms
2021-02-09 21:31:14,364 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 4.82ms
2021-02-09 21:31:25,693 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 5.03ms
2021-02-09 21:31:25,703 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 4.77ms
2021-02-09 21:31:36,809 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 4.96ms
2021-02-09 21:31:36,818 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 4.77ms
2021-02-09 21:31:48,040 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 5.06ms
2021-02-09 21:31:48,049 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 4.87ms
2021-02-09 21:31:59,119 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 5.10ms
2021-02-09 21:31:59,130 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 4.99ms
2021-02-09 21:32:10,360 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 4.96ms
2021-02-09 21:32:10,371 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 5.29ms
2021-02-09 21:32:21,603 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 4.98ms
2021-02-09 21:32:21,613 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 4.81ms
2021-02-09 21:32:32,853 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 4.95ms
2021-02-09 21:32:32,863 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 5.04ms
2021-02-09 21:32:44,121 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 5.06ms
2021-02-09 21:32:44,129 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 4.85ms
2021-02-09 21:32:55,374 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 5.02ms
2021-02-09 21:32:55,383 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 4.81ms
2021-02-09 21:33:06,609 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 5.04ms
2021-02-09 21:33:06,619 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 4.86ms
2021-02-09 21:33:17,874 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 5.12ms
2021-02-09 21:33:17,882 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 4.89ms
2021-02-09 21:33:29,111 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 5.02ms
2021-02-09 21:33:29,120 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 4.97ms
2021-02-09 21:33:40,375 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 4.90ms
2021-02-09 21:33:40,386 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 4.88ms
2021-02-09 21:33:51,622 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_notification (127.0.0.1) 4.95ms
2021-02-09 21:33:51,632 - tornado.access - WARNING - 404 GET /api/plugin/zbolt_octoscreen?command=get_settings (127.0.0.1) 4.92ms

Now looking up what tornado.access is and why it is not finding a page?  Is this because I don't have the plugin loaded and as yet I have not found that plugin :-(

tre4b, 2021-02-09:
And following that I figure out this IS because it is not finding the plugin and by searching for zbolt_octoscreen I found the plugin which could be loaded here

https://github.com/Z-Bolt/OctoPrint-Z-Bolt-OctoScreen/archive/master.zip

Loaded that and it is all working.  If you just add that to the end of the install info for OctoScreen it would be good.  Was a good learning experience though so you can just go ahead and close this now...

JeffB42, 2021-02-10:
Hi @tre4b , glad you got it working and got everything sorted out.  You have it working, but it seems like there are two issues & bugs here:
1) Using an Access Application key instead of a General API key, OctoScren doesn't pick up the API key, and one has to SSH in and edit the config file.  Darn, and I was going to remove loading the API key from the config file.  I'll look into this b/f I release 2.7.
2) When the OctoScreen plug-in isn't installed, it's crashing and causing a restart.  The warnings you listed in the posting two posts ago (the list of 404 warnings) are just warnings (or at least, should be) and not having the plugin installed shouldn't be causing a restart.  I normally run with the plugin installed, so I'll look into this as well.  I'll also reduce the level of those 404s for the plug in from warning to info, since it's not an error condition and shouldn't be even be a warning (or at least, it shouldn't be)

I'm going to open this back up, so I have something to track & make sure it's addressed (you know, out of sight, out of mind)

Thanks for posting all this and thanks for troubleshooting the issues.  If you find any other issues, let me know.

Jeff

tre4b, 2021-02-10:
Thanks Jeff,  can confirm all aspects seem to function correctly once I have the Plugin loaded.   When I installed this 0.18 version it forced me to add a user name.  I think there is no longer an option to not use the Access API and this could be what is causing the getting the key from the file to not work?  Maybe it no longer exists in the way it did for the older versions or something.

Interesting thing too was that I set it up on my desk and everything worked fine (once an API key was added) giving me just the basic info.  I then moved it to the printer and plugged in the USB and the blanking issue then started to happen.  Suggests that the crashing issue may be related to something aware the printer is there and trying to talk to it.

What is also interesting is once this had happened unplugging the printer did not correct this.  the Pi remained on the login screen and the basic OctoScreen never appeared again until I rebuilt.  That may help you figure out where the challenge is I hope.

cwgismo, 2021-02-19:
Not shore if this the same issue "tre4b" is having but it appears to be very similar to the issue I am having. randomly my Octoscreen disperses and quickly flashes (what I believe to be) the raspberry shell. Sometimes it happens on its own other times it happens when select something on the screen. it will occasionally happen during a print however it dose not effect the print in any way. I have also tried Octoprint in safe mode and it still happens. I am uploading a video to show what I am referring to as well as my log file. I am using a Raspberry Pi 2 model B v1.1 with the official Raspberry Pi 7" Touch Display

https://user-images.githubusercontent.com/78937118/108568451-183bb280-72d8-11eb-9cc7-10294b40b62a.mp4
[octoprint(4).log](https://github.com/Z-Bolt/OctoScreen/files/6013322/octoprint.4.log)

If anyone has any idea what is happening hear and how I can fix it I would really appreciate the help.

tre4b, 2021-02-20:
@cwgismo  I did think originally that this could be something to do with the Login options in the raspi-config menu.  When I enabled this though after that OctoScreen does not run any longer.  I suspect enabling the option in raspi-config knocks something from Octoscreen out of a script somewhere.   If you are brave you could try it though.

JeffB42, 2021-02-21:
> If anyone has any idea what is happening hear and how I can fix it I would really appreciate the help.

What errors do you see in your log file?
https://github.com/Z-Bolt/OctoScreen/wiki/FAQ#q-how-do-i-enable-logging--why-would-i-want-to-turn-logging-on

cwgismo, 2021-02-21:
> > If anyone has any idea what is happening hear and how I can fix it I would really appreciate the help.
> 
> What errors do you see in your log file?
> https://github.com/Z-Bolt/OctoScreen/wiki/FAQ#q-how-do-i-enable-logging--why-would-i-want-to-turn-logging-on
my log is set to debug and i don't see any error. the only instance of error I see is:
Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking

tre4b, 2021-02-22:
@cwgismo   I think this is a different issue to my original one.  However I am having a similar issue. and have reported this on a different thread [https://github.com/Z-Bolt/OctoScreen/issues/234#issuecomment-782771823](url)

The logs that Jeff is after are the octoscreen ones not the octoprint main logs I suspect this link will help you turn that on

https://github.com/Z-Bolt/OctoScreen/wiki/FAQ#q-how-do-i-enable-logging--why-would-i-want-to-turn-logging-on

I have done this already and sent it in for my situation, but more information cannot hurt.

JeffB42, 2021-02-27:
@cwgismo what errors are you seeing in you (OctoScreen) log file?
https://github.com/Z-Bolt/OctoScreen/wiki/FAQ#q-how-do-i-enable-logging--why-would-i-want-to-turn-logging-on

cwgismo, 2021-03-01:
> @cwgismo what errors are you seeing in you (OctoScreen) log file?
> https://github.com/Z-Bolt/OctoScreen/wiki/FAQ#q-how-do-i-enable-logging--why-would-i-want-to-turn-logging-on

This is my octoscreen loge file hopefully you can find something useful.
[octoscreen_logfile.docx](https://github.com/Z-Bolt/OctoScreen/files/6064839/octoscreen_logfile.docx)

martinmillicay, 2021-03-04:
Hello, sorry I don't know English but I can tell you that I have the same problem. Everything perfect until I connect my printer, it detects the serial connection but then the raspberry command line appears. Could you help me with this problem by explaining me with simple steps to avoid confusion? Thanks!

tre4b, 2021-03-04:
@martinmillicay 

I suspect you have a problem I have.  You need to make sure you load the plugin into OctoPrint as well as get OctoScreen running.  Go into octoprint and add a plugin.   You will see you can add one from a URL.   Go to this box and put in

https://github.com/Z-Bolt/OctoPrint-Z-Bolt-OctoScreen/archive/master.zip

This will load the Octoscreen plugin and then you should find that OctoScreen will work.  @JeffB42  has flagged this crash as a bug however it will work if you make sure the Plugin is loaded as detailed above.

martinmillicay, 2021-03-04:
Good thing, the problem has been solved.  Thanks a lot!  Regards.

JeffB42, 2021-03-27:
This has been fixed in 2.7.0, which was just released.

closed 2021-03-27 by JeffB42