Z-Bolt/OctoScreen #221: Can't install Octoscreen , service does not start

Snip3rita, 2021-02-11:
Hi everyone,
I'm using Octoprint since 2 years now and I wanted to install Octoscreen on a 3.5 inch GPIO display.

I followed this guide:
https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

But I keep getting an error when trying to install Octoscreen, I tried different guides and read the other issues but I never found the same problem. Can anyone help me out? If I go on, ignoring the problem, the service won't start and the display keeps blinking on the console.

> pi@octopi:~ $ sudo dpkg -i octoscreen_2.6.1_armhf.deb
[sudo] password for pi:
(Reading database ... 58081 files and directories currently installed.)
Preparing to unpack octoscreen_2.6.1_armhf.deb ...
Unpacking octoscreen (2.6.1-1) over (2.6.1-1) ...
Setting up octoscreen (2.6.1-1) ...
Job for octoscreen.service failed because the service did not take the steps required by its unit configuration.
See "systemctl status octoscreen.service" and "journalctl -xe" for details.

This is the status:

> pi@octopi:~ $ systemctl status octoscreen.service
● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: activating (start) since Thu 2021-02-11 12:00:32 GMT; 1s ago
 Main PID: 20188 (xinit)
    Tasks: 2 (limit: 877)
   CGroup: /system.slice/octoscreen.service
           ├─20188 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor
           └─20189 /usr/lib/xorg/Xorg :0 -nolisten tcp -nocursor

Feb 11 12:00:32 octopi xinit[20188]:         Before reporting problems, check http://wiki.x.org
Feb 11 12:00:32 octopi xinit[20188]:         to make sure that you have the latest version.
Feb 11 12:00:32 octopi xinit[20188]: Markers: (--) probed, (**) from config file, (==) default setting,
Feb 11 12:00:32 octopi xinit[20188]:         (++) from command line, (!!) notice, (II) informational,
Feb 11 12:00:32 octopi xinit[20188]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Feb 11 12:00:32 octopi xinit[20188]: (==) Log file: "/var/log/Xorg.0.log", Time: Thu Feb 11 12:00:32 2021
Feb 11 12:00:32 octopi xinit[20188]: (==) Using config file: "/etc/X11/xorg.conf"
Feb 11 12:00:32 octopi xinit[20188]: (==) Using config directory: "/etc/X11/xorg.conf.d"
Feb 11 12:00:32 octopi xinit[20188]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Feb 11 12:00:33 octopi xinit[20188]: modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/5.10.14+

Thank you

markeby, 2021-02-13:
It looks to me that your X11 configuration isn't working and isn't the fault of OctoScreen.  You might want to check your X11 setup and even see if you can do a simple test https://avikdas.com/2018/12/31/setting-up-lcd-screen-on-raspberry-pi.html

Snip3rita, 2021-02-14:
Could it be 'cause I'm using an old Raspberry 1 B+ ??
I'm not able to use octoscreen, touchui or octodash..I even changed display from a 3.5 GPIO to a 5" Hdmi..
I followed almost any guide..

JeffB42, 2021-02-14:
That could be part of the issue.  Remember, it's running not only OctoScreen but OctoPrint as well.  From what I've heard, a RPi1 (and probably a RPi 2) or a RPi Zero are too under powered to run OctoPi (let alone OctoPi + OctoScreen).  I suggest getting an RPi4.  The 2 GB version will work, as well as the 4 GB and 8 GB models, but I think 4 or 8 GB is overkill.  I normally run OctoPi + OctoScreen on a 2 GB RPi 4, and a Hyperpixel display.

Jeff

Snip3rita, 2021-02-14:
I will try with a 3 b+ (1Gb) next week , a friend has one..
I see lot of people using it..

thebeline, 2021-03-13:
@Snip3rita - What were your results?  Is this resolved?

Snip3rita, 2021-03-13:
> @Snip3rita - What were your results? Is this resolved?

Yeah sorry..the problem was my hardware..works fine with 3b+ . Thank you

JeffB42, 2021-03-13:
Good to hear.  BTW, 2.7.0 will be released either this weekend or next weened, so keep an eye out.

closed 2021-03-13 by JeffB42