Z-Bolt/OctoScreen #224: Random terminations while connected to a remote octorpint server

hawkeyexp, 2021-02-13:
I'm using 2 pi's (gen 4) one as server and one is a terminal (rpi display / case including pi etc.) and the terminal has installed octoscreen accessing the 1st pi which is running octoprint and connected to the printer.
I noticed random crshes (black screen and octoscreen will restart) - i optimized wifi by disabling power management of wifi which made it more stable but the crashes are still there. Possible a crash which is caused by delayed response from main system by a low timout setting in code ?

JeffB42, 2021-02-16:
> Possible a crash which is caused by delayed response from main system by a low timout setting in code ?

It's possible.
* Are you running 2.6.1?
* What model of RPi are you running?  You mentioned gen4, do you mean a Raspberry Pi 4b? https://www.raspberrypi.org/products/raspberry-pi-4-model-b/
* What do you see in OctoScreen's log file?

hawkeyexp, 2021-02-16:
I have tested with 2.6.1 and cause i had the effect i tested with dev branch both with same problems.
I checked the logs but didn't notice errors.
Both systems are rpi4b - octorpint server 4gb ram / octoscreen system 1gb ram

hawkeyexp, 2021-02-17:
I had a deeper check but the only thing i could identify is xserver is crashing.

Sidenote: with latest commits after "minor format change to open.svg" octoscreen won't start anymore ...

JeffB42, 2021-02-20:
@hawkeyexp I added some additional guards (and a lot of additional logging) and your issue might be fixed with the 2.7 release. I'll probably release 2.7 in the next few weeks so keep an eye out for it.

hawkeyexp, 2021-02-21:
ok i will give the dev branch a try and do a compile :-)

JeffB42, 2021-03-27:
@hawkeyexp I just release 2.7.0.  I'm not 100% positive it fixes your issue, but 2.7 does fix a few bugs, like restarting if the printer isn't on/isn't connected.  Try 2.7.0 out and let me know if this is still an issue.

JeffB42, 2021-05-01:
@hawkeyexp Any update on this?

hawkeyexp, 2021-05-02:
I'm sorry to say cause i liked the system but i have switched to octodash since a while which does a good job and i started loving the plugin integrations for filament manager etc and the easy way to configure the custom buttons which is more clear than in octoscreen.

closed 2021-05-03 by JeffB42