Z-Bolt/OctoScreen #225: Alpha / Beta release states

markeby, 2021-02-13:
Please tag the dev branch and/or publish when you believe the version in development is eligible for Alpha and Beta testing.  We know it may not be stable but are willing to contribute to the release process in a positive way.  I do some dev work and could help at those stages.

JeffB42, 2021-03-02:
I try to keep the active dev branch stable, and I usually don't push a commit until the feature I'm working on is either finished or mostly finish, so if curious I say just pull the code and run it.  I'm almost finished with 2.7 so if you want to take a look, I'd say that now is a good time to check it out.

thebeline, 2021-03-13:
@JeffB42 - I know that you are effectively the sole dev for this, but I agree with @markeby, I would love access to a bleeding-edge branch, not just for testing, but also as we discussed, for submitting PRs to.

JeffB42, 2021-03-20:
@markeby it's not tagged, but if you pull the sources from 2.7.0-dev and build, it's almost ready for release.