Z-Bolt/OctoScreen #226: Can't control secondary tool ( dual extruder)

misterfab31, 2021-02-14:
Hi,

I installed recently octoscreen 
Using raspberry pi 3b+, octoprint configured, Bigtreetech pitft 5", octoprint plugin installed also.

Printer is controlled correctly (Tevo Tornado) for normal printing.
I launched several prints during the last two days and had no problem with Octoscreen.

But I have a dual extruder on my printer (two stepper motors and single nozzle)

I activated "tool changer" in the plugin menu of octoprint, but I don't have the possibility on the touchscreen to select which extruder to control, I can only control the default tool#0.

Is there something I missed or is it a bug?

JeffB42, 2021-02-14:
I think it's a bug.  I'm guessing that within OctoPrint, in the printer profile you created, you set the profile to 2 extruders, and checked "Shared Nozzle" on.   The logic in places where the nozzle is heated or moved is:
if Shared Nozzle is true
    display only one nozzle
else
  display nozzles based on count of tools list returned from API

The actions in the Filament screen follows this logic.  It appears that for actions that are extruder-related, the tool list should also be used.   I'll look this over and fix it in  the Filament panel.

Jeff

misterfab31, 2021-02-15:
Thank you for your answer Jeff

closed 2021-02-16 by JeffB42