Z-Bolt/OctoScreen #228: Custom Commands Don't Support Multiple Commands

thebeline, 2021-02-16:
The Custom Command list in the `Control` panel passes on Custom Commands that do not contain a `command` or `script` property:

https://github.com/Z-Bolt/OctoScreen/blob/7ef3c0277b084a9730210922b790372a2ec2c5d8/ui/ControlPanel.go#L84

However, by default, the `Control` Button *also* supports the `commands` property:

https://github.com/Z-Bolt/OctoScreen/blob/7ef3c0277b084a9730210922b790372a2ec2c5d8/uiWidgets/ControlButton.go#L51

It would make sense to include that as a property for adding.

**Additionally:**  Currently the upstream OctoPrint Custom Commands parser fails pretty bad with parsing `input` properties, which kind of saves OctoScreen, in that it does not support `input` either.  It may be beneficial to add a check that the Custom Command *does not* include any `input` properties, just in case the upstream library gets that fixed.

And finally, although I am not sure if these properties are supported upstream or not, but it may be beneficial to *also* filter out any Custom Controls with the properties `javascript` and/or `enabled`, as both of those properties are designed to run in-browser.  Maybe even `regex` and `template` too, as those are designed for rendering in-browser, and I don't see where there are provisions to render output to OctoScreen at this time...

JeffB42, 2021-02-20:
Thanks for reporting this @thebeline , I'll try to look into this in the 2.8 release.

JeffB42, 2021-03-27:
This was fixed in 2.7.0, thanks to @thebeline

PR #263 (2021-03-27): Bug Fix: CustomControls with multiple command lines and Control/Command Confirm handling.
Well, I WAS testing this.  The build is fine, and it does populate the list with the Custom Control (NEW! NEAT!), but clicking does all of nothing.  So I am working to sort that out now.

Resolves #264 
Resolves #228
closed 2021-03-27 by JeffB42