Z-Bolt/OctoScreen #229: FR: Support Display of M117 and M118 for Status Update or Message Display

thebeline, 2021-02-16:
It would be incredibly useful if OctoScreen supported updating, somewhere, the contents of `M117` and/or `M118` messages.  I understand there are a few things at play here, so let me address them one at a time:

**Where:**  I am not certain.  Certainly, if the printer receives these *while* printing (as part of a G-Code file that is actively executing), somewhere on the Printing Panel would be ideal.  However, it is also possible that these codes will be sent/received during ANY G-Code execution, so perhaps a simple (scrolling, if need be) overlay at the top or bottom of the screen in any other panel.

**How:** It does occur to me that OctoScreen does not receive messages directly from serial, so this would have to be a feature active in the OctoScreen OctoPrint Plugin, that reports these messages to OctoScreen.

It also occurs to me that OctoPrint does NOT receive `M117` messages, as it is not the 'LCD' per-se.  I have yet to test this, but I am guessing that OctoPrint **may** receive these if it is connected to the controllers on-board LCD serial bus, in which case I will have to see "how" and in "what manner" it receives these to have any suggestions there.

As for `M118`, I would imagine we would handle these if they are prefaced with the `echo:` command (`echo:Hello World`).

**When:** Well, that would be up to OctoScreen and it's preferences.  Off the top of my head, to simplify things, I would say that the OctoScreen OctoPrint Plugin just caches the last message it has received, and reports that to the OctoScreen GUI on request, with an `E-Tag`, perhaps, so that OctoScreen can determine if the message has updated since the last request (OctoScreen requests the message while providing an `E-Tag` and the OctoScreen OctoPrint Plugin either sends a new message if the message in cache does not match the `E-Tag`, or nothing if the current message in cache *does* match the `E-Tag`.

I think this **can** be on-request only, and I **don't** think it needs to maintain a list of all messages.  I think the OctoScreen Interface would only need to display the latest one, and not worry about any it may have missed.

If there is an option to "read" or "dismiss" the currently displayed message (in the event that it is in a pop-up on any screen other than the Print Panel), then it just closes, and doesn't worry about any others (like displaying a stack).  If a new one comes up, it just displays the next one.

**Determining what TO and NOT to cache/display:** As I write this, I am guessing it may be beneficial to just have a place to define a `regex` for Status Message Parsing in the OctoScreen OctoPrint Plugin to parse received messages.

This would make handling ANY received message fairly trivial (weather it is an `M118` **with** or **without** the `echo:` prefix, or whatever format the `M117` code generates).

Perhaps a few different Regular Expressions:
 - Blacklist (messages to straight ignore, such as `~^(ok|echo:busy:|(\w(X|T):[0-9]+.+))$~i`)
 - Emergency Statuses (whatever the regex for those are, I would have to check)
 - Standard Statuses (whatever those may be, `M118` with the `echo` flag take the form of `~^echo:(.+)$~i` where the captured group is the message)

**Future Features:** The groundwork for this would also allow indicating to OctoScreen if the printer is busy (disabling Control and Command buttons until the Printer is Idle).

This would be handled by caching the latest of either `~^echo:busy\w*(.+)$~i` (which indicated the Printer is Busy, and with What (capturing group), or `~^\w*ok\w*$~i` which indicates that the printer has completed the previous command.

**API:** Perhaps an endpoint in the OctoScreen OctoPrint Plugin that returns a json array of both the Job Status (busy/idle) as well as the Job Message ("Processing", which maybe it always is) as well as the latest `M118`/`M117` message, and the level of that message (Info, Error, etc, based on the regex match).

If there are already ways in which this does just this, perhaps some documentation and extension to allow a little more flexibility or for people to code against it.

Maybe I am completely missing the mark here and just spent all this time writing this for nothing...  POSSIBLE....

JeffB42, 2021-02-20:
> Maybe I am completely missing the mark here and just spent all this time writing this for nothing... POSSIBLE....

LOL... I've been there and can relate.

This is a great write up though, and let me try to address some of the points you brought up.

First is the issue of how to get this information.  You are correct, in that OctoScreen doesn't directly communicate with the printer, and instead, all communication is done via REST APIs that OctoPrint eposes.  If you want to take a look, the docs for their APIs are at https://docs.octoprint.org/en/master/api/, and it most likely would be in the Job APIs, https://docs.octoprint.org/en/master/api/job.html.  I haven't looked into it, but I don't think M117 and M118 status is sent back.

Another option (but this is grasping at straws) is the DisplayLayerProgress plug-in or the M117PopUp plug-in might expose a REST API.  If that's the case, what I could do is call their API... if it's not installed I just get back a 404 error (or one of the other errors) and just ignore it, but if I get back a 200 response, I could read & parse the results and then display the status/message.  But this would only be >>IF<< one of those plug-ins actually exposes a REST API.

Second, where to place the information.  The printing screen is already pretty filled up.

[image]

If the M117 or M118 messages are expected to not be very long (single row, maybe 30 or 40 characters max) I might be able to add an additional row of text (a label control) underneath the  time left label.

> ...so perhaps a simple (scrolling, if need be) overlay at the top or bottom of the screen in any other panel.

This is not possible.  The OctoScreen application is written in GoLang and uses GTK for the UI system, and unfortunately, GTK is not robust enough for that.

thebeline, 2021-02-21:
HEY YO! @JeffB42 - Thanks for even considering it!

I have been looking some after I wrote that Scotch-inspired feature request, and I think we hit about the same thing!  `DisplayLayerProgress`!

They provide a WebSocket API, which I think would work well for this (also a standard REST API, but I think the WebSocket [if activated] would be the best bet).

As for scrolling, you are right, that is SUPER over-kill anyway.

As for where: Well, on the Print Panel, I would assume, from a UX perspective, just under the File Name, OR just under the progress bar.  Frankly, as OctoScreen can effectively be an enclosure panel replacement, it is imperative that whatever may be placed on the OEM LCD display ALSO be presented in OctoScreen.

As for where to display elsewhere, that is kind of up to you, but here me out:

I have some custom actions (that may be executed from the Controls panel, when that gets sorted), and I have a fairly custom Auto Bed Level script, that DOES get executed via another panel.  I use `M117` and `M118` extensively to give me an idea of where in the process I am (bed pre-heat procedure, if I am waiting on the enclosure temp, other sensors, etc).  It would be SWELL if notifications had a place in the rest of the UI, lower or top left corner, I don't know, but it isn't a deal breaker.

On the Print panel, however...  I think it is essential.

The `DisplayLayerProgress` plugin is used and flagged as a dependency for a few other plugins, so I think it is a safe bet to code against that. I don't think it is grasping at straws at all.

WebSocket is fairly easy to code against, and I am fairly certain Go has significant support for it.  Let me do a quick little looksie.

thebeline, 2021-02-21:
[GoRilla](https://github.com/gorilla/websocket/blob/master/examples/echo/client.go) looks to provide a comprehensive WebSocket implimentation, with a client library and example (linked) as well.

I have not delved into your code much, but I IMAGINE you are using threads, yeah? Spawn a thread once you've confirmed OctoPrint is running to connect to the DLP websocket, and basically...  If it fails, sure, if it connects...  MONEY

I have NO experience in Go, so...  I'm no help I am afraid.  :-(  But this would be dope...

thebeline, 2021-02-22:
Or maybe a slight layout shift:

[image]