Z-Bolt/OctoScreen #231: Octoscreen doesn't work

Ducati87, 2021-02-17:
Hello guys, I'm trying to setup octoscreen no my rasp3b but when it boot up, octoscrren doesn't appear. 
This is a log of service activity:
```

pi@octopi:~ $ x11
-bash: x11: command not found
pi@octopi:~ $ sudo systemctl status octoscreen
[sudo] password for pi:
● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor prese
   Active: activating (start) since Wed 2021-02-17 09:29:51 GMT; 785ms ago
 Main PID: 2286 (xinit)
    Tasks: 2 (limit: 2062)
   CGroup: /system.slice/octoscreen.service
           └─2286 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocurs

Feb 17 09:29:51 octopi xinit[2286]: (EE)
Feb 17 09:29:51 octopi xinit[2286]: Fatal server error:
Feb 17 09:29:51 octopi xinit[2286]: (EE) no screens found(EE)
Feb 17 09:29:51 octopi xinit[2286]: (EE)
Feb 17 09:29:51 octopi xinit[2286]: Please consult the The X.Org Foundation supp
Feb 17 09:29:51 octopi xinit[2286]:          at http://wiki.x.org
Feb 17 09:29:51 octopi xinit[2286]:  for help.
Feb 17 09:29:51 octopi xinit[2286]: (EE) Please also check the log file at "/var
Feb 17 09:29:51 octopi xinit[2286]: (EE)
Feb 17 09:29:51 octopi xinit[2286]: (EE) Server terminated with error (1). Closi
lines 1-18/18 (END)
● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: activating (start) since Wed 2021-02-17 09:29:51 GMT; 785ms ago
 Main PID: 2286 (xinit)
    Tasks: 2 (limit: 2062)
   CGroup: /system.slice/octoscreen.service
           └─2286 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor

```

hawkeyexp, 2021-02-17:
Have you used the public 2.6.1 or compiled from dev 2.7 ?

Ducati87, 2021-02-17:
I used the precompiled .deb procedure.

hawkeyexp, 2021-02-17:
Hm should work out of the box on an octopi image - octopi 0.18 deb 2.6.1 install with dpkg and reboot - all working usually

JeffB42, 2021-02-20:
@Ducati87 This looks like the X11 libraries aren't installed.  Either that or the display isn't set up correctly.  Take a look at https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display and https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

Ducati87, 2021-02-21:
I solved the problem formatting the SD and installing all over again.

closed 2021-02-21 by Ducati87