Z-Bolt/OctoScreen #232: Debian logon screen

ingo916, 2021-02-18:
How do I prevent this from happening?
[image]

JeffB42, 2021-02-18:
Well, for starters, you can't use Debian - you have to use an OctoPi distro.

ingo916, 2021-02-18:
Lol, I did use the OctoPi distro. After a few reboots OctoScreen loads. 

Linux octopi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Feb 17 18:51:31 2021

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

------------------------------------------------------------------------------
Access OctoPrint from a web browser on your network by navigating to any of:

    http://octopi.local
    http://

https is also available, with a self-signed certificate.
------------------------------------------------------------------------------
OctoPrint version : 1.5.3
OctoPi version    : 0.18.0
------------------------------------------------------------------------------

pi@octopi:~ $

JeffB42, 2021-02-18:
Out of the box, OctoPi doesn't have Debian installed.  I haven't tested w/0.18 yet... unless something changed in 0.18, one needs to explicitly run `/home/pi/scripts/install-desktop` in order to install Debian.  I guess I'll need to update the setup docs and mention to not run the install-desktop script.

> After a few reboots OctoScreen loads.

If things are working, then great.  If you run into any issues withe display though, you might want to consider either removing Debian, or install again and reflash the SD card.  OctoScreen uses X11, as does Debian, and they don't coexist.

closed 2021-02-27 by JeffB42