Z-Bolt/OctoScreen #234: HDMI Screen woes

tre4b, 2021-02-20:
Two issues this time.  I decided to buy an HDMI touch screen to use rather than the WaveShare type I was using.  It was supposed to give better response./

On first install everything seemed good however every so often the system would drop back to Pi console then restart OctoScreen.  I tried going into raspi-config and setting the option to login automatically as the Pi user.  This made octoscreen not appear at start.

I tried removing and reinstalling OctoScreen but this did not make it start with reboot.

Using SSH and  ps -ef | grep "octoscreen" I have confirmed the process is there and running after reboot.

If I do "sudo systemctl restart octoscreen.service  it will run again, still doing the original alternation with the console screen.

I also tried stopping and starting with the same overall effect as restart.  Here's some command lines

`pi@octopi:~ $ ps -ef | grep "octoscreen"
pi        3608  3483  0 18:40 pts/0    00:00:00 grep --color=auto octoscreen
pi@octopi:~ $ sudo systemctl stop octoscreen.service
[sudo] password for pi:
pi@octopi:~ $ ps -ef | grep "octoscreen"
pi        4103  3483  0 18:40 pts/0    00:00:00 grep --color=auto octoscreen
pi@octopi:~ $ sudo systemctl start octoscreen.service
Job for octoscreen.service failed because the control process exited with error code.
See "systemctl status octoscreen.service" and "journalctl -xe" for details.`

Can't understand why doing this would make octoscreen vanish never to show again unless manually restarted.

JeffB42, 2021-02-21:
> On first install everything seemed good however every so often the system would drop back to Pi console then restart OctoScreen

Two things:
First, this sounds like OctoScreen is crashing and then restarting.  Turn logging on and set the level to debug, reboot, run, and then look at the log file for errors.
https://github.com/Z-Bolt/OctoScreen/wiki/FAQ#q-how-do-i-enable-logging--why-would-i-want-to-turn-logging-on

Second, in a week or two, I will be releasing version 2.7.0.  it fixes some bugs which caused OctoScreen to crash, so your issue may have been fixed.

tre4b, 2021-02-22:
Hi Jeff,

Done that and rebooting now (work was boring anyway ;-) ).   

I booted as normal and octoscreen failed to start.  Checked the logfile at that point and it was blank, so no error being registered.   I do wonder if somehow using Raspi-config to set the login option changed something that directed the display away from octoscreen some how.  If I get a chance I will redo from scratch to see if without this nothing breaks.

Next I restarted octoscreen with sudo systemctl restart octoscreen.service

That then gave me this logfile.

[logfile.txt](https://github.com/Z-Bolt/OctoScreen/files/6021041/logfile.txt)

This has been running for about 5 mins and doing it's thing with restarting so should show the issue somewhere if it is in there.

MajdiQahtani, 2021-02-22:
`Two issues this time. I decided to buy an HDMI touch screen to use rather than the WaveShare type I was using. It was supposed to give better response./`

Hi tre4b

I know I am not providing any help to you in this post, but I think you can help me.

Would you please help explain to me how you got the Waveshare LCD to work with OctoScreen? It is driving me crazy.

I have the WaveShare 4-inch (800x480) HDMI touch LCD which I tried installing on both a Raspberry Pi 3 and 4, and I just cannot have it display the OctoScreen flash screen. I get the error "no screens found(EE)" in the log file /var/log/Xorg.0.log.

I managed to have the LCD work with the graphical touch display if I install the full Raspberry OS with desktop environment by following the instructions on Waveshare website. However, the OctoScreen graphical environment is not working. I only have the black screen command prompt when I reboot.

This is the process I followed:
   1. Flashed the SD card with a fresh OctoPi image 0.18.0 

   2. Inserted the Waveshare setting in the config.txt file, as listed in the Waveshare website.

   3. Installed the GTK+3 graphic environment required to run OctoScreen using the command: sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils

   4. Installed the Waveshare drivers from git clone https://github.com/waveshare/LCD-show.git

   5. Installed OctoScreen using the commands:
      - wget https://github.com/Z-Bolt/OctoScreen/releases/download/2.6.1/octoscreen_2.6.1_armhf.deb
      -  sudo dpkg -i octoscreen_2.6.1_armhf.deb

Both of the Raspberry Pi's 3 or 4 that I am using go to the command line black screen after trying to reboot. I have the raspi-config set to log on with the pi account to the command line.

I would appreciate your guidance on the settings you have in config.txt and the  /usr/share/X11/xorg.conf.d/40-libinput.conf , and the sequence of installing the OctoScreen and the Waveshare drivers, and any other settings I am missing.

Thanks.

tre4b, 2021-02-22:
hi @MajdiQahtani I just used the excellent document provide by the guys here: 

https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

With the caveats

1)  You need to create an Octoprint API Key in the Octoprint UI.  Then add that to the key in the /etc/octoscreen/config file.  The thing to do with automatically picking up a key does not seem to work on 0.18

2)  You also need to load the pluggin for Octoprint in Octoprint.  Do that through the addins and then put this in the HTML box 
https://github.com/Z-Bolt/OctoPrint-Z-Bolt-OctoScreen/archive/master.zip

Other than that you are set to go.  I did every reboot in there and did not shortcut anything.  I did this install twice and both times it worked.  Do not ever use raspi-config to set autologin though as that breaks it and the only way I found to get it working again was to reinstall the second time.

MajdiQahtani, 2021-02-22:
Thank you very much tre4b. 

I searched GitHub many times and for some reason I did not see the link you sent me. Thank you again.

MajdiQahtani, 2021-02-23:
> If I do "sudo systemctl restart octoscreen.service it will run again, still doing the original alternation with the console screen.

Hi tre4b,

You may know this already, but I understand that the OctoScreen altenates with the console command screen briefly if the 3D printer is not connected through the USB port.

Also, to ensure that the OctoScreen boots every time, I added the command `systemctl start octoscreen` at the bottom of the /etc/rc.local.

This may help with your problem.

BTW - the octoscreen is working with my Waveshare 4-inch LCD finally. Thank you for the guidance.

tre4b, 2021-02-23:
My printer is connected,  I suspect it is something to do with having tried to use raspi-config.  My own fault the touch screen was not working because I had inverted it without noticing and the software was not using the same orientation.  I thought maybe it was because of the login so I tried raspi-config then after that Octoscreen will not display on its own and needs me to start it with systemctl restart.    I think I can fix it with a rebuild and not doing raspi-config however it would be better to have a fix for it for octoscreen as using raspi-config should not break the screen and it will only come up again if not fixed.  I'm hoping the new version due soon will fix it.

JeffB42, 2021-03-27:
@tre4b I just released 2.7.0.  I suppose you could try it and see if you issues are fixed, but it sounds like you have a library issue.  After installing 2.7.0, If you still run into issues, I suggest starting from scratch:
1. using a different SD card, reflash OctoPi
2. install and set up OctoPrint
3. install and set up OctoScreen
4. don't forget to consult  https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display and https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen
...and if you still have issues, then 5. turn on logging and look for errors in the log file.

tre4b, 2021-03-29:
Hi Jeff,

I will give it a try with just upgrading,  just in case it works and would work for others.   If that fails then I will go ahead and rebuild.  I've been using octoprint in basic settings anyway and I can always backup and recover the settings.

Let's hope the easy way works 😉  I've swapped my screen out for one with HDMI now anyway so that should make things easier.

Trev

closed 2021-05-01 by JeffB42