Z-Bolt/OctoScreen #235: Octoscreen Service Masked

DavidMilroy, 2021-02-21:
I've set up a new Raspberry Pi 4 with 4GB.  Installed Octoprint without any problems.  Added a Raspberry 7" touchscreen to the Pi 4 and it is working properly.  When I start it up, Octoprint starts, I can see if from my PC browser.  However, on the 7" touchscreen, I only have the terminal interface.  At one point when I was setting it up, I did see the Octoscreen icons, but now they are gone.

When I check to see if the Octoscreen service is running I enter:
     systemctl status octoscreen.service

It returns:
   octoscreen.service
       Loaded: masked (Reason: Unit octoscreen.service is masked.)
       Active: inactive (dead)

I am really a newbie with Linux, Raspberry PI, and everything else.

Any help anyone could offer would be appreciated greatly!!

JeffB42, 2021-02-27:
I don't know, but you could try uninstalling OctoSrcreen and then reinstalling it.  Or run through the setup steps: 
https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display
https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

Also turn OctoScreen's logging on and look for errors:
https://github.com/Z-Bolt/OctoScreen/wiki/FAQ#q-how-do-i-enable-logging--why-would-i-want-to-turn-logging-on

JeffB42, 2021-03-27:
@DavidMilroy any update on this?

JeffB42, 2021-03-30:
This issue was also reported here: 
https://github.com/Z-Bolt/OctoScreen/issues/149

I'm not familiar with masking, @thebeline does this sound familiar?  Any ideas as to what's going on?

@DavidMilroy it sounds like something is misconfigured pretty badly.  Maybe you used `sudo` when you didn't need to, or maybe you didn't use `sudo` when you needed to (maybe, I'm just guessing here).  What I would do is take a new SD card (or reuse your current one), flash it with the latest OctoPi, install and configure OctoPrint, and then run through the steps at https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display and 
https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

If after having installed clean and you get this same error, I did a google search and came across
https://medium.com/@codingmaths/service-masking-in-linux-f7265d9b2181 and https://unix.stackexchange.com/questions/308904/systemd-how-to-unmask-a-service-whose-unit-file-is-empty and they might help

thebeline, 2021-03-30:
Curious.

This should be able to be resolved by running the install process again.  Follow the [UpdateInstructions](https://github.com/Z-Bolt/OctoScreen/releases/tag/v2.7.0), you will not loose your settings.

If you still have issues, try running `systemctl unmask octoscreen.service` prior to following the upgrade instructions.

That being said, @JeffB42: The main `README` contains a link to the last version (`2.6.1`), and is also broken.

If this does not resolve the issue, please let us know.

closed 2021-05-24 by JeffB42