Z-Bolt/OctoScreen #237: FR: Remove individual Print buttons from file list, add Print button to bottom

thebeline, 2021-02-22:
The individual Print buttons on the file list don't really provide a good UX.  I propose removing the buttons and adding a Print button to the bottom of the screen, furthest to the right.

Coupled with #236, one would select the file they wish to print, have a visual confirmation that they have identified the correct file, and then press the Print button.

As it stands, with all of the Print buttons in a row, ham-handed individuals may go to print one file, but there is no confirmation that they selected the correct one.  Highlighting the selected row is a start, but also consolidating the "Print" action to a single button further solidifies the UX.

Additionally, this may allow to shrink the rows, which may solve the off-screen behavior the bottom buttons currently exhibit.

Additional Feature: Perhaps replace the Print buttons on each row with a smaller "Details" button.  This panel could display further details about the file, including full name, time, layers, and any other details that OctoPrint may make available.  You could also have a Print button on this screen, so a user could execute the print from here (you would still want a select/print option in the current panel, to minimize required screens for an action).

JeffB42, 2021-02-27:
I'm going to keep the printer buttons there and not change the layout, but the upcoming 2.7 release has a lot of changes in the file select panel:
- preview thumbnails (when available) are displayed
- selecting anywhere in the row (not just the printer icon) will select the print and invoke the confirmation dialog box
- the confirmation dialog box now has a white border, making it easier to see and stand out
- background color of each row alternates
- selecting a print changes the background color

closed 2021-02-27 by JeffB42

thebeline, 2021-03-02:
Love it!  Thanks.

thebeline, 2021-03-02:
@JeffB42 - When will that be being released?  Do we have a timeline?