Z-Bolt/OctoScreen #238: FR: Customize the RPi Boot Process for cleaner look.

thebeline, 2021-02-23:
It would be cool to include a script or something part of the install process that customizes the default RPi boot process to pull the whole look together, since these are typically dedicated devices, and by and large will probably be mounted to the printer.

After some quick searching, it seems this is fairly straight forward:

```
$ echo "disable_splash=1" >> /boot/config.txt
$ sed -i 's/console=tty1/console=tty3/g' /boot/cmdline.txt
$ cat << EOF
logo.nologo
consoleblank=0 loglevel=1 quiet
EOF >> /boot/cmdline.txt
$ apt install fbi
$ cat << EOF
[Unit]
Description=Splash screen
DefaultDependencies=no
After=local-fs.target

[Service]
ExecStart=/usr/bin/fbi -d /dev/fb0 --noverbose -a $SPLASH_IMAGE_PATH
StandardInput=tty
StandardOutput=tty

[Install]
WantedBy=sysinit.target
EOF >> /etc/systemd/system/splashscreen.service
$ systemctl enable splashscreen
```

Roughly.  This is what I gather from [this blog post](https://yingtongli.me/blog/2016/12/21/splash.html) from 2016 (un tested at the moment).

In the same post it does seem you could even do animated, by (after generating the correct `*.bmp` files):

```
$ git clone https://github.com/alukichev/bannerd.git
$ cd bannerd
$ make
$ sed -i "s/ExecStart=.+/ExecStart=/bin/sh -c '/usr/local/bin/bannerd -vD /$SPLASH_FRAMES_PATH/\*\.bmp'/g" /etc/systemd/system/splashscreen.service
```

I am going to test this all out here shortly, we'll see just how "easy" this really is, but I imagine it's fairly close, and would be SUPER chill.

JeffB42, 2021-02-27:
Thanks for contributing @thebeline .  Looks like you posted an updated issue on this at https://github.com/Z-Bolt/OctoScreen/issues/240 so I'm going to close this issue.

thebeline, 2021-02-28:
This is actually a different topic.  #240 is about installing on 3.5" screens, this is more general.  I'll polish this up a tad too.  :-D

thebeline, 2021-03-05:
So, I got this running on my pi with a 3.5" screen.

Notes:
- On the 3.5" display, there is nothing I can seem to do about the white blanking for about 5 seconds prior to loading the splash.
- There is a 2 second black screen blank between the splash and OctoScreen displaying, which is annoying, but is what it is.
- The animated splash seems to wait for Xsession before it runs, so it runs at the same time as OS, and runs on top of it unless it is killed (keeps looping).
  - This can be mitigated by calling `pkill -TERM bannerd`, but as this seems to execute about tandum with OS, it is of little use to us (in my testing).

I used this image:
[image]

This could be included in the distribution, OR could be managed by the OctoPrint plugin to allow users to upload a custom image.

I am going to do a from-scratch configure on another machine now, I will post how smooth it goes in a moment.

thebeline, 2021-03-05:
Worked a treat!!

Note: `fbX` is the final Frame Buffer (if there are multiple `/dev/fb*`, it is the last one).  On my 3.5" screen, there is an `fb1`, which is what I needed to set this to.  On my other install, it is driven by `DSI` so there is only one Frame Buffer, `fb0`, and that is what THAT needed to be set to.  I will need to extend the script to determine this.