Z-Bolt/OctoScreen #240: Update: LCD35 Install Process simplified

thebeline, 2021-02-24:
I'll clean this up, and maybe draft up a shell script that can be tracked, but I ran a BUNCH of tests, and can confirm that you can install OctoScreen in, what is essentially, two commands.  It's still "a bunch of commands" but below is just the command line version of what would really be in a shell script.

I tested this on a clean OctoPi, as well as what I can only describe as an "abused" OctoPi (decidedly NOT fresh, loads of things already on it, as well as aborted/orphaned installs of a handful of things), and it works.

Of Note: the ONLY `reboot`s "needed" were the one following `LCD35-show` (and that is only because the script FORCES a reboot) OR after installing and configuring OctoPi (the final `reboot now` below could be replaced with `service octoscreen rstart` and everything is G2G).

*Also* of Note: Unfortunately, the OctoScreen `deb` automatically starts the OctoScreen `service`, which in and of it's self isn't "bad."  However, what **is** bad is that the OctoScreen does NOT release the `start` if it can not load the API Key.  Which causes `service octoscreen start` to hand until it times out.  This seems unacceptable, as, on a fresh install, it is highly likely that there won't be an API Key to load, and this causes supreme issue...  Perhaps release the `start`...  That'd be swell...

```
sudo sh -c 'apt-get update && apt-get install -fy libgtk-3-0 xserver-xorg xinit x11-xserver-utils git build-essential xorg-dev xutils-dev x11proto-dri2-dev libltdl-dev libtool automake libdrm-dev' && \
git clone https://github.com/waveshare/LCD-show.git && \
chmod -R 755 LCD-show && cd LCD-show/
```

Instructions to check driver (if this were a script, we could prompt the user to select the correct file).

```
sudo bash ./LCD35-show
```

Automatically Reboots (GRRRRR)

```
sudo sh -c 'cp /boot/config.txt /boot/config.txt.bak && \
sed -i -re "s/^\w*(hdmi_cvt.*)$/#\1/g" /boot/config.txt && \
echo "hdmi_cvt=800 533 60 6 0 0 0" >> /boot/config.txt' && \
cd ~ && git clone https://github.com/ssvb/xf86-video-fbturbo.git && \
cd xf86-video-fbturbo && \
autoreconf -vi && ./configure --prefix=/usr && make && \
sudo sh -c 'make install && \
cp xorg.conf /etc/X11/xorg.conf' && \
cd ~ && wget https://github.com/Z-Bolt/OctoScreen/releases/download/2.6.0/octoscreen_2.6.0_armhf.deb && \
sudo sh -c 'dpkg -i octoscreen_2.6.0_armhf.deb && \
cp /etc/octoscreen/config /etc/octoscreen/config.bak && \
echo "Please log in to your OctoPrint Instance and retrieve an API Key for OctoScreen to use" && \
read -p "OctoPrint API Key: " OCTOPRINT_APIKEY && \
sed -i -re "s/^\w*(OCTOPRINT_APIKEY.*)$/OCTOPRINT_APIKEY=$OCTOPRINT_APIKEY/g" /etc/octoscreen/config && \
sed -i -re "s/^\w*(OCTOSCREEN_RESOLUTION.*)$/OCTOSCREEN_RESOLUTION=800x533/g" /etc/octoscreen/config && \
reboot now'
```

The above `reboot now` is not required if we have already done the reboot after `LCD35-show`, and this could be replaced with `service octoscreen restart`.  

If I were writing a script, and not doing all of the same-command stuff above, I would add more prompts, some options to exit or continue (or skip interactive mode entirely), and re-start checks (to skip portions of the script that had already executed).

I've used some fairly robust interactive `bash` script frameworks in the past, I will see if I can dig one up and make a script for this.

Leaving this here as a not to myself...

thebeline, 2021-02-26:
Eh, it's a little prettier.  Atleast it gives some feedback and such now...

```
#!/bin/bash

sudo sh -c 'apt-get update && apt-get install -fy libgtk-3-0 xserver-xorg xinit x11-xserver-utils git build-essential xorg-dev xutils-dev x11proto-dri2-dev libltdl-dev libtool automake libdrm-dev'

[[ -d LCD-show ]] || \
        { echo "Cloning LCD-show"; git clone https://github.com/waveshare/LCD-show.git; }

[[ -d LCD-show ]] || \
        { echo "Unable to clone LCD-show"; exit; }

{ chmod 755 LCD-show/* && cd LCD-show/; } || \
        { echo "Failed to set permissions and enter LCD-show."; exit; }

echo "Executing LCD-show"
systemd-inhibit sudo bash ./LCD35-show

echo "Configuring the Boot Config."

sudo sh -c 'cp /boot/config.txt /boot/config.txt.bak && \
sed -i -re "s/^\w*(hdmi_cvt.*)$/#\1/g" /boot/config.txt && \
echo "hdmi_cvt=800 533 60 6 0 0 0" >> /boot/config.txt' || \
        { echo "Failed to configure Boot Config."; exit; }

cd ~

[[ -d xf86-video-fbturbo ]] || \
        { echo "Cloning FBTurbo."; git clone https://github.com/ssvb/xf86-video-fbturbo.git; }

[[ -d xf86-video-fbturbo ]] || \
        { echo "Unable to clone FBTurbo."; exit; }

cd xf86-video-fbturbo

{ autoreconf -vi && ./configure --prefix=/usr && make; } || \
        { echo "Failed to build FBTurbo."; exit; }

sudo sh -c 'make install && \
cp xorg.conf /etc/X11/xorg.conf' || \
        { echo "Failed to Install X11 or copy X11 Configuration"; exit; }

cd ~

echo "Installing OctoScreen"

{ wget https://github.com/Z-Bolt/OctoScreen/releases/download/2.6.0/octoscreen_2.6.0_armhf.deb && \
sudo sh -c 'dpkg -i octoscreen_2.6.0_armhf.deb && \
cp /etc/octoscreen/config /etc/octoscreen/config.bak && \
echo "Please log in to your OctoPrint Instance and retrieve an API Key for OctoScreen to use" && \
read -p "OctoPrint API Key: " OCTOPRINT_APIKEY && \
sed -i -re "s/^\w*(OCTOPRINT_APIKEY.*)$/OCTOPRINT_APIKEY=$OCTOPRINT_APIKEY/g" /etc/octoscreen/config && \
sed -i -re "s/^\w*(OCTOSCREEN_RESOLUTION.*)$/OCTOSCREEN_RESOLUTION=800x533/g" /etc/octoscreen/config && \
reboot now'; } || \
        { echo "Failed to something something something OctoScreen"; exit; }

```

Took a look at the OctoDash (gasp, shock, awe) install script, and it is 750 lines of interface code, and like maybe 30 of the actual install...  Well, it is bash after all...  I hate bash...

Anyway, I mention that install script because I think I am going to whip up a pretty.  I'll poke it at some point...

(Side Note: I did try OctoDash...  I hated it.  Ironically, I have done some things to try to fubar my Pi, but trying to peal OctoDash out of it and get OctoScreen running again...  That was a special hell...)

JeffB42, 2021-02-27:
Thanks for contributing and helping out @thebeline .  One request that I have is to add the version number to be passed into the script.  At the moment the script is hard coded to 2.6.0 (and the latest is 2.6.1 BTW).  I'm going to follow the same naming convention in future releases, so anywhere there is "2.6.0" should be changed to a parameter that's passed in.  Bonus would be to run a regex on the parameter at the beginning of the script and make sure it's three numbers and two periods, and follows the pattern of "num.num.num"

Some other things to keep in mind are:
* This obfuscates the install process.  OctoScreen attracts many users, with different levels of experience in Linux, so try and think of ways the script might fail, and try and make it bullet proof.

* The script is specific to systems with displays needing the LCD35 drivers.  That's fine, and I'll eventually add it to the 3.5" guide (https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen), but it's specific to 3.5" 480x320 displays which use the LCD35 drivers.

* OCTOPRINT_APIKEY.  This is a tricky one.  The global key is defined in OctoPrint's config.yaml file, and OctoScreen tries to read it if its there.  So, strictly speaking, it isn't required to be defined in `/etc/octoscreen/config`.  However, OctoPrint is moving away from a global API key and to individual application keys, and they are stored in a different file.  I haven't added support for this yet, but that's something to also keep in mind.

* Uninstall script.  If you provide an install scrip, and uninstall script would be helpful as well.  I don't know if it would need to uninstall everything, but at the least it should probably run `sudo dpkg -r octoscreen`

> I did try OctoDash... I hated it. Ironically,

I like OctoDash.  It think it, OctoScreen, and the other ones each have their own pros & cons.  For me, I liked OctoScreen's UI and layout better (I'm not the original developer), and it seemed to have been designed from the beginning to run on a small display.  Not to knock OctoDash, but there are some UI elements that are hard to tap w/o tapping a nearby control, and not everything displays large enough to read, like the error messages and confirmation dialog boxes.  It's been a while since I looked at OctoDash so maybe things have changed, but I think everyone should try it out and decide which is best for them.

> ...trying to peal OctoDash out of it and get OctoScreen running again... That was a special hell..

Yea, I don't recommend that.  I've added a note in the setting up guide about supported distros and supported hardware (No, it will not run on a Pi-Zero!) and I should probably add a section explicitly stating how nothing else 
(like the Debian desktop, or OctoScreen and OctoDash side-by-side) should be installed.

Anyway,
1) Thanks for the help!
2) I look forward to the next version, and when ready I'll add it to the 3.5" display install guide
2) Glad you are up an running with OctoScreen (and happy 3D printing!)

Jeff

thebeline, 2021-03-05:
OctoDash Removal catch:

When converting from (either a standard Desktop, or some other GUI such as OctoDash) you *must* clear the default Multi-User target.  This can not be done via the `raspi-config`, and it must be done manually by:

```
sudo rm -rf /etc/systemd/system/default.target
```

Reboot, and if OctoScreen is installed, it will start correctly at boot.

thebeline, 2021-03-05:
Note on the above install script:

I have removed the `inhibit` as it does seem that the reboot with regards to `LCD35` **IS** required (it seems this is the only one, however).

This will necessitate this being two scripts if there is going to be an "Install" script.

Further, I have added the above mentioned `default.target` removal.  Frankly, this blocks the correct operation of OctoScreen anyway, so it should probably be a part of the Deb package.

thebeline, 2021-03-16:
@JeffB42 - I just read through your comments a bit slower...  :-D  So, here we go I guess (when I refer to "Installer" I mean #247):

- Version Numbering: I handled that in the installer.  It just installs the most recent version.  I can't see why we would want people selecting a specific version, but that is certainly an option I could add if you feel it is a requirement.
- Bullet-proof: Well, back to the installer, it's about as bullet-proof as the instructions as they stand.  I suppose I could add more debug statements, but I am not sure for what..  ;-)
 - With regards to LCD install, I haven't written that in yet, but I think I am setting up good personal patterns to make it the kind of bullet-proof we'll need.  We'll see, when I integrate it.
- Specificity: I was going based on what the LCD 3.5 guide was instructing, and as that was needed by, and written specifically to, 3.5 inch LCD users, I thought the specificity was acceptable.  I will remember to add some selection options into the script to select the driver-type.
  That being said, I imagine we don't want anyone installing on SMALLER than 3.5 inches, and on my 5 inch screen I didn't need to do anything...  But...  Sure.  What criteria/limitations do you think the handler should have?
- Looks like `2.7.0-dev` auto-selects the "hard-coded" API Key in the OctoPrint Config.  So this is probably not required.
  Regardless, that is more of a step of the Installer, so I will integrate that into that particular Task at some point.
- Uninstall: YUP, already handled in the installer (well, "Manager" I suppose, at that point).
- OctoDash/et al: I have determined the fix for this issue, I will be submitting a PR with the fix in the DEB install script shortly.

thebeline, 2021-03-16:
#sidenote: This being my second Issue on this repo...  you probably didn't expect at the time that you'd be hearing from me so much all of a sudden...  ;-)  Cheers!

thebeline, 2021-03-16:
#noteforself: This should probably be titled `--drivers` or something, dunno.  Either way, process:
- Driver Install after dependencies, before `build` or any OctoScreen Install or Remove stage.
- Sanity checks to determine if (LCD32, etc) already installed:
 - `/boot/overlays/waveshare*.dtbo` exists and matches expected (check scripts to confirm)
 - `/usr/share/X11/xorg.conf.d/99-fbturbo.conf` exists
 - `/usr/share/X11/xorg.conf.d/99-calibration.conf` exists
 - Also of note, the build directory will exist and not be empty.
- If already installed do not re-install unless `--force` or ask User (allows for executing the same command line after the restart that occurs).
- Do the same for FBTurbo
 - `/etc/X11/xorg.conf` exists and contains Driver `fbturbo`
- No re-install same as above
- Continue with next tasks (`build`/`install` etc)

thebeline, 2021-03-19:
Added to #247