Z-Bolt/OctoScreen #241: Octoscreen Autoconnect Printer

tibmeister, 2021-02-28:
Need to have option to disable this so that firmware updates can occur, and we can stop filling logs when the printer is turned off but Octoprint is left to run

JeffB42, 2021-03-02:
You can stop the service by running `sudo systemctl stop octoscreen`

tibmeister, 2021-03-02:
Yeah, I know how systemd works.  I don't really want to have to SSH in to stop a service, to then be able to use the WebGUI to upload the firmware.  Plus, at least one instance when I was troubleshooting I did stop the unit (octoscreen) and it showed as stopped, but there was an instance still running when I did a "ps -ef | grep octos".  I ended up disabling the unit and rebooted the pi to validate that octoscreen was for sure doing this odd autoreconnect, and also filling up the logs like crazy.  Without some logrotate function, it would be bad to fill the partition up with log entries when the printer is turned off for possibly a week or two.
Anyway, having an option to opt out of having it perform autoconnect every couple of seconds.

thebeline, 2021-03-13:
@JeffB42 - I do see that `2.7.0-dev` has a `Retry` feature, I have not delved into the code for this yet, but as a short response: does OctoScreen initialize the Printer Reconnect automatically anymore?  Or was this removed?

IMHO, it would seem like this is not the responsibility of OctoScreen (in an automatic sense), and default behavior would be to just query OctoPrint to see if it has connected.

@tibmeister - You seem knowledgeable, could you build from branch `2.7.0-dev` and see if this behavior persists?

tibmeister, 2021-03-15:
@thebeline It will be a couple of days; I have several prints I need to finish and don't want to muck around with a working config until that's done.

seppelicous, 2021-03-21:
Hey guys,
I do have the same issue in my setup. After printing, my TP Link socket turns off but octopi stays alive to e available for new prints. It automatically turns the socket on & off. So that's a wonderful automation.

Now since I installed Octoscreen the log file and my push notifications on iOS (using Octopod) are full of "error connection" messages. I figured out, that this stops when I switch off Octoscreen. Glad to see, that this behaviour stops with the 2.7 release.

I would like to test the branch - but since I am not a tech nerd in raspbian and git... how can I update the installation with this 2.7.0 branch? I installed from deb file before.

I appreciate your help - came from Octodash and Octoscreen is so much more useful for me!

seppelicous, 2021-03-28:
2.7.0 still has this bug. My Ender is switched off with a TP smart plug and only alive on demand, while OctoPi is always on.
There is no issue when Octoscreen is disabled, but when Ocotscreen is active and Octoprint has no active connection, it throws errors. 

Because of this issue I went back to Octodash. Not I once again have to switch back do OctoDash.  It is not usable for me

thebeline, 2021-03-28:
This patch was not included in the 2.7 release.  I am sorry this gave you issues.

IBNobody, 2021-04-19:
> Hey guys,
> I do have the same issue in my setup. After printing, my TP Link socket turns off but octopi stays alive to e available for new prints. It automatically turns the socket on & off. So [...]

I think I am having the same issue as you (on 2.7.2 now). I recently upgraded my setup to a Sonoff S31 smart outlet running Tasmota, and I am using the Tasmota plugin for OctoPrint. I was puling my hair out trying to figure out why OctoPrint kept trying to reconnect to my printer because the connection attempts would turn on my smart outlet.

A workaround for me was to disable the Tasmota plugin's option to turn the printer on during a connect attempt. Now, OctoScreen can try all it wants to reconnect to a powered-off printer without being able to turn the printer on.

Still, it would be nice if OctoScreen would leave a disconnected printer disconnected (and maybe have a button to connect)...