Z-Bolt/OctoScreen #242: Advise: Package process to build .deb for testing of commits and branches

thebeline, 2021-03-05:
I am looking to submit some changes, but want to test my changes prior to submitting PRs.  I want to test the process end-to-end, so I want to actually build a `.deb` and install my changes with that package.

Can you please advise on the correct build process?  I have never packaged a `deb` before, OR worked with Go, so...  WEEEEE

Thank you!

thebeline, 2021-03-06:
So, I think I figured it out (ish).  On top of all of the other things I had to install due to `fbturbo` and whatnot, I had to install `docker.io`, and run `$ sudo make build`.

Problem is (**not for anyone else wanting to do the same**): Running this on a RaspberryPi...  Is...  Like...  **SLOW**

Despite the fact that I am compiling for Arm (the Pi), can I run this on an x86 machine does anyone know?  Because...  I don't think I want to go through this every time I want to test a build...

Also... Just watching this...  (it hasn't finished) it looks kind of like it is fetching the data from the GitHub repo...  Which kind of makes me think it is not going to compile my local changes...

thebeline, 2021-03-06:
Also, I got some "useless" dependency notifications, just saying.

JeffB42, 2021-03-06:
>...want to test my changes prior to submitting PRs

Yes, that's mandatory. Don't be like this guy:

[image]

> Can you please advise on the correct build process

I think the build steps are already documented, https://github.com/Z-Bolt/OctoScreen#install-from-source

You'll need this path:
`~/go/src/github.com/Z-Bolt/OctoScreen`
...and then clone the repo

`sudo rm build/stretch/*`
`make build`

and then

`sudo dpkg -i {name of deb packages that was built}`

> I have never packaged a deb before, OR worked with Go, so... WEEEEE

If you have a Pluralsight subscription, they have several courses on GoLang.  If not, there are resources out there on how to learn GoLang.  If you've used C, C++, C#, or Java before, there are some similarities.  There's also a lot of differences too, and I suggest taking some time to learn the language and get familiar with it.  I'm not a fan of GoLang, but that's what the project is written in, and I'm not a fan of NIH or rewriting working code when rewiting doesn't provide much value.  OctoScreen also use GTK and the GTK wrapper package for Go - https://github.com/gotk3/gotk3.

> Running this on a RaspberryPi... Is... Like... SLOW

I have two systems I run and test on: a RPi3B + 3.5" display, and a RPi4 + 4" HyperPixel.  My RPi4 is MUCH faster at building and running.  If you are going to do any development, I suggest doing it on a RPi4.

A few things to mention:
* Only target PRs for dev branches, never master.  There's a article on this in the wiki.
* I'm about to release 2.7.0.  Adding anything new right now will be too chaotic and disruptive, so go ahead and create PRs, but they won't go in until 2.7.1 or 2.8.0.
* Before creating a PR, open either a bug ticket or a feature request ticket first, and document the issue and any proposed changes.
* I'm a stickler on syntax and formatting.  While I don't agree with every coding style that's used in the project, I like consistency.  PRs that have changes just for the sake of formatting will be asked to remove the unnecessary formatting changes.  Before submitting a PR, be sure to read through the code and get a sense of the existing style, structure of the code, and the architecture.
* I'm going to write up a guide on how to submit a feature request and how to document and create UI prototypes for the FRs, but haven't had the time yet.

Also, few other things:
1) Thanks for the help.  I look forward to others working on this project.
2) I'm about to release 2.7.0.  If you are pulling and building the code, what would be a help right now wold be to pull the 2.7.0-dev branch, build it, and test it.  I test it myself, though having more eyes on it wouldn't hurt.
3) I want to keep OctoScreen printer-agnostic.  When requesting a FR, try to keep that in mind, and ask yourself if the FR would benefit only a select few users, or all OctoScreen users.  I've gotten a few Prusa-specific and Prusa MMU-specific FRs, so I might create a panel specific to that machine, but I'd like to keep it generic and have universal appeal.
4) I'm spending too much time replying to issues - each weekend I'm spending 2-3 hours just replying back to posts here.  Because of this I've put limits/boundaries and I only look at issues on the weekends, so if days go by w/o a reply from me, it's because I'm offline (so to speak)
5) Along the lines of #1, #2, and #4, if you are looking to help, actively monitoring issues and assisting in helping others get configured and up and running is a big help and always appreciated.

Again, thanks for any help,

Jeff

thebeline, 2021-03-06:
All noted.  I will begin eating my own dogfood and building and coding against the `dev` branches.

I did figure out the building process.  I was an idiot for not reading the readme before asking that...  It does look like the directory you mentioned gets created automatically, and I have built `master` a few times now with success, so, awesome.

On that, when submitting a PR for something that may or may not be on the roadmap (and therefore may not have an associated `dev` branch) where should I target the PR?

I will try to adhere to the existing style, and I will try not to submit PRs that are frivolous.

As for printer-specific...  Yeah, I don't operate that way.  I very much subscribe to generalization (perhaps too much).  I won't be too specific like that I think you will find.

As I get more comfortable in the code, I will ask fewer things and leaving a PR to sit for some time is fine by me.  I hope to be useful.  Enjoy your weekend.

JeffB42, 2021-03-07:
> I did figure out the building process.

Excellent, glad it's building and working out

> I was an idiot for not reading the readme before asking that

No worries, I've done that myself.

> where should I target the PR

Normally, you would look for the latest dev branch, which is 2.7.0-dev at the moment.  However, I'm close to releasing it, so it would be better to hold off.

> As I get more comfortable in the code, I will ask fewer things

No worries about asking questions.  If you run into any issues, or have any other questions, just let me know.

Jeff

thebeline, 2021-03-13:
This is resolved.

closed 2021-03-13 by thebeline