Z-Bolt/OctoScreen #245: Tool Change Button not displayed or missing and Filament change length is not applying

FloxRoxX, 2021-03-08:
Hi, at first i want to thank you for such a great and good looking software!

If this post should belong to the Octoprint Plugin i can reopen it there again.

Iam using Octoscreen with a dual extruder setup. Now iam missing the Toolchange button for changing the toolhead.
This button is crucial for working with the display to get controll of the second extruder ect... 
Now i know i can enable it through the plugin in octoprint but it doesnt show up! Oh and also the filament change length inside the plugin is not working. I already tried rebooting the pi but it did not showed up or the filament change lenght applied...

Any suggestions ?

JeffB42, 2021-03-08:
Version 2.7 is going to be released soon, and it has some fixes for filament extruding, retracting, loading, and unloading.  Also, there wasn't work done on a tool change function, but I aded support for multiple extruders.  When you define you printer settings in OctoPrint, when OctoScreen runs, if there are multiple extruders, there's now a "extruder select" button in the Filaments panel.  Hopefully this will give you want you're looking for.  If not, ping me after 2.7 releases and let me know what's needed.

JeffB42, 2021-03-27:
2.7.0 was just released.

closed 2021-03-27 by JeffB42