Z-Bolt/OctoScreen #246: Octoscreen not loading

kgauler, 2021-03-10:
Hallo,
so far I could install 3,5"-display, GUI and OctoScreen, but restarting Raspberry I get only the Octoscreen-icon and the message "Loading..."

With systemctl status octoscreen.service I get the following error message, but I have no idea how to solve it:

● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: activating (start-post) since Wed 2021-03-10 17:42:48 GMT; 9s ago
 Main PID: 922 (xinit); Control PID: 952 (bash)
    Tasks: 29 (limit: 1938)
   CGroup: /system.slice/octoscreen.service
           ├─922 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor
           ├─923 /usr/lib/xorg/Xorg :0 -nolisten tcp -nocursor
           ├─936 /usr/bin/OctoScreen
           ├─952 /bin/bash /etc/octoscreen/disablescreenblank.sh 0
           └─953 sleep 10

Mar 10 17:42:49 octopi xinit[922]: time="2021-03-10T17:42:49Z" level=debug file=main.go func=main.readConfig line=211
Mar 10 17:42:49 octopi xinit[922]: time="2021-03-10T17:42:49Z" level=debug msg="OctoScreen - entering main.setApiKey()" file=main.go func=main.setApiKey line=128
Mar 10 17:42:49 octopi xinit[922]: time="2021-03-10T17:42:49Z" level=debug msg="main.setApiKey() - APIKey is empty, now using cfg.API.Key" file=main.go func=main.setApiKey line=132
Mar 10 17:42:49 octopi xinit[922]: time="2021-03-10T17:42:49Z" level=debug msg="main.setApiKey() - APIKey is \"A64---1CC\"" file=main.go func=main.setApiKey line=141
Mar 10 17:42:49 octopi xinit[922]: time="2021-03-10T17:42:49Z" level=debug msg="OctoScreen - leaving main.setApiKey()" file=main.go func=main.setApiKey line=144
Mar 10 17:42:49 octopi xinit[922]: 2021/03/10 17:42:49 logger.SetLogLevel() - the log level is now set to: "error"
Mar 10 17:42:49 octopi xinit[922]: 2021/03/10 17:42:49 main.SetLogLevel() - logLevel is now set to: "error"
Mar 10 17:42:50 octopi xinit[922]: time="2021-03-10T17:42:50Z" level=error msg="ui.verifyConnection() - s.Current.State is IsOffline, and (ConnectRequest)Do(UI.Client) returned an error: \"unexpected status code: 403\"" file=logger.go func=utils.LogError line=86
Mar 10 17:42:50 octopi xinit[922]: time="2021-03-10T17:42:50Z" level=error msg="Widow resize went past max size.  allocatedWidth:480 allocatedHeight:320 sizeWidth:500 sizeHeight:390" file=ui.go func=ui.New.func1 line=75
Mar 10 17:42:50 octopi xinit[922]: time="2021-03-10T17:42:50Z" level=error msg="Widow resize went past max size.  Target width and height: 480x320" file=ui.go func=ui.New.func1 line=80

Many thanks in advance
Klaus

thebeline, 2021-03-13:
What is the make and model LCD you are using?

Do you get terminal output on the display on startup (or anything?)

Please provide the output from the following commands:
```
cat /etc/X11/xorg.conf
cat /boot/config.txt
cat /etc/octoscreen/config | grep RESOLUTION
```

thebeline, 2021-03-13:
My thought is that you have OctoScreen's resolution set higher than the Frame Buffer or `cvt` resolution, which would cause the system to go bonkers.  In which case, find the highest resolution that is not `#` commented out from `xorg.conf` (which typically doesn't contain resolutions) or `boot.txt` (if you know what you are looking for), and set `OCTOSCREEN_RESOLUTION` to that value in `/etc/octoscreen/config`.

Again, if that doesn't work, send over the results of those commands and I will have a think on it...

JeffB42, 2021-03-13:
@kgauler I see two errors in the log you provided:

#1
Mar 10 17:42:50 octopi xinit[922]: time="2021-03-10T17:42:50Z" level=error msg="ui.verifyConnection() - s.Current.State is IsOffline, and (ConnectRequest)Do(UI.Client) returned an error: "unexpected status code: 403"" file=logger.go func=utils.LogError line=86

Is OCTOPRINT_APIKEY set?
is OCTOPRINT_HOST set?
is your printer on and connected?

#2
Mar 10 17:42:50 octopi xinit[922]: time="2021-03-10T17:42:50Z" level=error msg="Widow resize went past max size. allocatedWidth:480 allocatedHeight:320 sizeWidth:500 sizeHeight:390" file=ui.go func=ui.New.func1 line=75
Mar 10 17:42:50 octopi xinit[922]: time="2021-03-10T17:42:50Z" level=error msg="Widow resize went past max size. Target width and height: 480x320" file=ui.go func=ui.New.func1 line=80

You have the issue where your screen isn't set up - it doesn't have enough pixels to display the UI, and the "window" is expanding.  Did you look over these set up guides?

https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display
https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

thebeline, 2021-03-14:
@JeffB42 - Thank you for the notes here.

thebeline, 2021-03-16:
> Mar 10 17:42:50 octopi xinit[922]: time="2021-03-10T17:42:50Z" level=error msg="ui.verifyConnection() - s.Current.State is IsOffline, and (ConnectRequest)Do(UI.Client) returned an error: "unexpected status code: 403"" file=logger.go func=utils.LogError line=86

`403` is `Forbidden`, which indicates we are hitting a valid host, but the API Key **is** wrong.  We can not determine anything further than that because the API connection has failed.  Check your API Key and restart.

> Mar 10 17:42:50 octopi xinit[922]: time="2021-03-10T17:42:50Z" level=error msg="Widow resize went past max size. Target width and height: 480x320" file=ui.go func=ui.New.func1 line=80
>
> You have the [...]

I don't completely understand this behaviour.  I am currently working on a branch that automatically determines the attached screen resolution and uses that (triggered by setting `OCTOSCREEN_RESOLUTION` to `AUTO`). @JeffB42 - Do you know of any reasons this could result in an error-state?

JeffB42, 2021-03-20:
> Do you know of any reasons this could result in an error-state?

Not from a screen resize, no.  I think they are two separate issues.

@kgauler How's it going?  Any luck? Make sure the following are set:

Is OCTOPRINT_APIKEY set?
is OCTOPRINT_HOST set?
is your printer on and connected?

...if you still have issue, turn on logging, set the level to debug, and look over the log file.

closed 2021-04-17 by JeffB42