Z-Bolt/OctoScreen #248: 2.7.0-dev Bug: OctoPrint Files Screen exits OctoScreen

thebeline, 2021-03-12:
Version: 2.7.0-dev (as of this issue)
Source: Compiled locally
System: Pi 4
OS: Ubuntu 10, Buster

Behavior:
Entering `Prints > OctoPrint` quits to login shell, restarts OctoScreen.

Reproduce: 
Well, I guess it isn't that simple.
- Print
- OctoPrint
- Back
- Back (to home)
- Repeat 2 more times
- Exits to login shell and restarts OctoScreen

If I were to guess, I would guess memory leak (the list of files of parsed data is not freed, and just stacks up until we fault out...  :-(

thebeline, 2021-03-13:
As mentioned in #249: This doesn't trigger like this on one of my smaller builds.

Also, it occurs to me that you probably want a log...

JeffB42, 2021-03-13:
I've not been able to reproduce this.  A few comments:

> 2.7.0-dev Bug: OctoPrint Files Screen exits OctoScreen

Darn, I was going to release 2.7.0 this weekend.

> Print
> OctoPrint
> Back
> Back (to home)
> Repeat 2 more times
> Exits to login shell and restarts OctoScreen

Thanks for the great rerpo steps!  It helps me a lot when they are clear like this.

> Back (to home)

One slight clarification here.  'home' here is the IdlePanel.  (and is in /ui/IdleStatusPanel.go)  I only mention this because there is a Home button and a HomePanel (/ui/HomePanel.go) which is used for homing the printer's toolhead.  I knew what you meant though.

> OS: Ubuntu 10, Buster

Did you mean Debian?  Or did is this correct?  if Ubuntu, are you building on Ubuntu and then transferring the deb file to your RPi?  If so, I'm not sure that's valid.  I tried once building on my Mac.  When the deb file is built, the compiler targets the current CPU, which isn't 1:1 with the RPi.  If you are seeing issues and not building this way, then try building on your RPi4, installing the package that was generated, and they reproducing the issue.

I tried the steps you listed.  I also tried the following:
steps 1-4 several times
steps 1-4 several times, going slow
steps 1-4 several times, going fast (in case it was some sort of timing issue)
steps 1-4 several times, but selecting "SD Card" instead of "OctoPrint", with the SD card inserted in my printer
steps 1-4 several times, but selecting "SD Card" instead of "OctoPrint", with no SD card in my printer

So my question then is, if you do the following
- pull latest 2.7.0-dev branch onto your RPi4
- build on your RPir4
- and then install
...are you able to repro the issue?  I haven't been able to repro it yet.

thebeline, 2021-03-13:
> Ubuntu

Ignore me, I was being an idiot again:

```
pi@octopi-ctc:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

thebeline, 2021-03-13:
> So my question then is, if you do the following

Yes.

However, note, I have 7 GB of gcode files on that instance.  The debug line for the returned JSON is nearly 700k characters long.

JeffB42, 2021-03-13:
> Yes, However, note, I have 7 GB of gcode files

Yikes.  I can't repro that.  Please turn on logging, set the level to debug, and then post the log.

Also, four other things:
1) what is the file count on your SD card?
2) could you try with an SD card that only has, say, 10 gcode files on it?  Try to repro it.  If it doesn't repro, then try 100 files.  And if 100 works, try 1000 files.
3) I'd tell you to just move the files into a folder and just leave 10 or so files at the root location (so the root location on your SD card would have 10 files, and 1 folder with all of your other files), but Marlin and Rambo motherboards don't implement M20 very well - even if you send the command with the "no recursion" options, Marlin and Rambo still treat it as a flat file system and return EVRY file.  For more info, see https://github.com/Z-Bolt/OctoScreen/wiki/Version-2.7#sd-card-limitations
4) Are you familiar with Postman?  Postman is a really good tool for calling and testing REST APIs.  I can help you get set up if you haven't used i before.  I think you might like it.

Thanks!

Jeff

thebeline, 2021-03-13:
I call rest from the command line (see other comments)

None of my systems have SD. But i can move files around pretty easy in uploads.

I'll run some tests.

I thought I attached logs, no?

thebeline, 2021-03-13:
Hrm... Um... I mist have attached the logs to a different issue...

JeffB42, 2021-03-14:
@thebeline 
> could you try with an SD card that only has, say, 10 gcode files on it? Try to repro it. If it doesn't repro, then try 100 files. And if 100 works, try 1000 files.

nm, I was able to repro it on my RPi3b.  I don't think it's the amount (disk space) of all the files, I think it's the number.  I was able to get it to repo with 300 files.  Haven't narrowed down how many it takes to cause the crash, or narrowed down the issue yet, but wanted to let you know I can repro it.  Thanks for posting and letting me know about it.

thebeline, 2021-03-14:
Since it is an enter/exit/enter thing, knee-jerk says there is a mem leak.

thebeline, 2021-03-14:
Any idea why it doesn't log errors?  That is...  concerning... no?

JeffB42, 2021-03-14:
I have a fix and pushed the changes.  It was due to a thread taking too long.  Please pull the latest and try it out.  My SD card has somewhere on the order of 325 files on it.  (I just created a small g-code file and then duplicated it)

thebeline, 2021-03-14:
Compiling now.

thebeline, 2021-03-14:
\#RESOLVED

closed 2021-03-14 by thebeline