Z-Bolt/OctoScreen #249: 2.7.0 Feature Request

thebeline, 2021-03-12:
Can you give us a config setting to disable printing from SD?  I am using OctoScreen to replace the whole gambit, I don't have an SD reader attached.

thebeline, 2021-03-12:
Better yet, query the printer to see if printing from SD is enabled in the firmware (is that a thing?).

thebeline, 2021-03-13:
[`setenv DO_I_HAVE_AN_SD_SLOT=false;`]

So `SD Card` shows up, but...  surprise surprise, doesn't work. In fact, on my smaller build, while #248 doesn't trigger, hitting `SD Card` a whole buuuunch of times does...  Maybe a quick query to that API prior to out that screen?

I am thinking, managed by the OctoScreen Plugin:
- Option for default "Print From"

OctoScreen proper:
- Immediately after connect, query both endpoints (`cat` data to `/dev/null`, ie, just use `curl` in a shell, to keep memory managed)
- If one or the other doesn't succeed, remove it in Go memory
- Initial `Print` screen get's fed remaining options
  - If only one exists, forward to that screen (regardless of `default`)
  - If the selected `default` exists, forward to that screen
  - If no default selected, either display what it currently does (in build `2.7.0-dev`), or pick one or the other (you, not the UI at random, although that would be fairly amusing)
- Button in top-right for the whichever is not `default`, if it's available
  Or the select-from-many screen that is currently displayed.

ADDED BONUS:
- Support for readers or attached USB drives?

Frankly, ME, I will not likely have an attached drive of any kind, and as such, I don't know what kind of support OctoPrint has for that (which I imagine OctoScreen needs, so...).

If I were to guess, I would say that OP likely only supports the local drive, and the controller's SD (if enabled, which I explicitly disabled in all of my FWs).  That said, at the OP level, it would be swell to see them support attached USB drives, if they don't already.

**Big takeaway:** Don't make the user press more buttons than they have to.  Period.  Hitting that screen made me take a step back, super not what I wanted to see.

thebeline, 2021-03-13:
So I just confirmed that the API only supports Uploads or SD, which is fine,

Two options:
1. Quick shell-based curl and word-count, if `wc -w` is greater than 1, there are files available:
  `$ curl -s -H "X-Api-Key: $OCTOPRINT_APIKEY" $OCTOPRINT_HOST/api/files/local | wc -w`
2. Query the `printer/sd` API, a status of `false` indicates no SD Card is available:
  `$ curl -H "X-Api-Key: $OCTOPRINT_APIKEY" $OCTOPRINT_HOST/api/printer/sd`
  `{"ready":false}`

thebeline, 2021-03-13:
Side note: If `/api/printer/sd` returns `false`, there IS an API to attempt to `init` the SD Card, however, unless there is a plan to implement ALL of the SD functions (`init`, `refresh` and `release`) I would advise against implementing any of them (perhaps for a following release).  I don't know.

JeffB42, 2021-03-13:
> I am using OctoScreen to replace the whole gambit, I don't have an SD reader attached.

Just to be clear, this displays files on the SD card that's inserted in your printer's SD card reader, not an external SD card reader that's plugged into the RPi's USB port.   OctoScreen calls OctoPrint's `/api/files/` API to retrieve the list of files (see https://docs.octoprint.org/en/master/api/files.html#get--api-files-(string-location) as well as `ui/FilesPanel.go`.

thebeline, 2021-03-13:
> Just to be clear, this displays files on the SD card that's inserted in your printer's SD card reader, not an external SD card reader that's plugged into the RPi's USB port. OctoScreen calls [...]

Yes, sorry, I have figured out the API, and I know.

I have removed the SD card reader, as well as the original LCD.
[image]

OctoPi replaces the whole gambut.  This IS controlled exclusively by OctoPi.

thebeline, 2021-03-13:
The above photo is from several months ago, I am running `2.7.0-dev` as of last week.

JeffB42, 2021-03-13:
BTW, a draft of the 2.7 release notes are up and are at https://github.com/Z-Bolt/OctoScreen/wiki/Version-2.7

JeffB42, 2021-03-13:
> Can you give us a config setting to disable printing from SD? I am using OctoScreen to replace the whole gambit, I don't have an SD reader attached.

I'd prefer not to.  I don't like the Microsoft approach where every option is configurable, and adding the logic for this bypass clutters up the code.  There were several requests to have the ability to print from SD card so I added it, and the best impl seemed (to me) to have a page where the user selects the source first.

I'm going to keep it as-is for now and let the feature get some use by the users.  If I get other requests to do away with the source select page, I'll consider adding a config setting to not show it and default to using OctoPrint's file repository.

closed 2021-03-13 by JeffB42

thebeline, 2021-03-16:
#251