Z-Bolt/OctoScreen #257: LCD35 3.5 inch Waveshare35A screen rotation issue

eblieb, 2021-03-18:
I was able to rotate the screen properly using dtoverlay=waveshare35a:rotate=270 in config.txt but I cant get the touch panel to rotate and/or calibrate properly? Any suggestions?

thebeline, 2021-03-18:
When following the 3.5 LCD install instructions, did you use the correct parameter for rotation when installing the LCD35 drivers?  As I understand it, you need to specify the rotation at that time for the installation to configure the digitizer rotation correctly.

thebeline, 2021-03-18:
Personal note: Be sure to account for rotations when implementing #240

JeffB42, 2021-03-27:
@eblieb The only things to mention are:
1. With some displays, the screen display and the touch matrix are independent, so they each have to be configured.
2. There are two setup guides: https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display and https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen.  While I can't help trouble shoot this issue for you, those guides have a lot of good information in them.
3. Chris's video on setting up OctoScreen with a HyperPixel talks about screen rotation.  https://www.youtube.com/watch?v=OJ59hXSyBoI.  Although it's not the same screen, his video does talk about display rotation and touch matrix rotation, and you might find something helpful.

Please keep me posted, as I'd like to close out this bug.

closed 2021-04-17 by JeffB42