Z-Bolt/OctoScreen #274: Change Extruder not working in 2.7.0

FloxRoxX, 2021-03-28:
Hi,
just upgraded to the new 2.7.0 release and tried out the "change extruder" button.
Monitoring the terminal in octoprint shows that no T0 or T1 command is executed on pushing the button.
Also extruding after setting the extruder on the screen did not extrude from the second one.
So i assume the T0,T1,T2 commands missing.

JeffB42, 2021-03-28:
So, three things...
1) It doesn't explicitly send a T0 (or T1, etc...) command, and leaves it up to OctoPrint to handle the switching.  Or at least it's supposed to...
2) There's a bug.  In utils/filament.go there is Extrude() and the tool id is passed in ("tool0", "tool1", etc...) which comes from the selectExtruderStepButton.  Right now it does (pseudo code here):
```
Extrude() {
    CheckIfHotendTemperatureIsTooLow() // and prompt the user if it's too low
    SetFlowRate() // calls /api/printer/tool with the "flowrate" command
    call ToolExtrudeRequest API // calls /api/printer/tool with the "extrude" command
}
```

...but this isn't doing anything with the toolID.  Instead, it should be:
```
Extrude() {
    CheckIfHotendTemperatureIsTooLow() // and prompt the user if it's too low
    SelectTool(tool ID) // calls /api/printer/tool with the "select" command
    SetFlowRate() // calls /api/printer/tool with the "flowrate" command
    call ToolExtrudeRequest API // calls /api/printer/tool with the "extrude" command
}
```

I'll put out a 2.7.1 release probably next weekend. (I want to see if any more bugs are reported)

3) That being said, my plan is to use OctoPrint's tool APIs (https://docs.octoprint.org/en/master/api/printer.html) and not issue any specific gcode, like T0, and I'm hoping this will work (not owning a MMU myself, it's hard to tell).  I'm hoping that when I call `/api/printer/tool` with:
```
{
  "command": "select",
  "tool": "tool1"
}
```
...that this will work for the MMU.  If the standard APIs don't work, I'll have to rethink this feature.

JeffB42, 2021-04-03:
@FloxRoxX I just released v2.7.1.  It doesn't explicitly send T0 (or T1, T2, etc..) commands, but this version now sends a `SelectTool` command to OctoPrint before extruding, which hopefully is sufficient for the MMU.  If curious about the code, see `Extrude()` in `https://github.com/Z-Bolt/OctoScreen/blob/master/utils/filament.go`, at line 35.

I don't own a MMU so I can't verify this personally.  If this doesn't work, let me know.

closed 2021-04-03 by JeffB42