Z-Bolt/OctoScreen #275: OctoScreen 2.7.0 don't recognize the OctoScreen plugin 2.6.0 installed

JeffB42, 2021-03-28:
@NicoIT31 reported this over in https://github.com/Z-Bolt/OctoPrint-Z-Bolt-OctoScreen/issues/16 but I'm moving this ticket here.  

OctoScreen 2.7.0 doesn't seem to recognize that the plug-in is installed.  When you go to the System information panel, it says "OctoScreen plugin not installed".

I was able to reproduce the issue one one system, but it worked fine in another, and my guess this is due to a timing issue (not fun to track down).  I will be working on this bug this week, and hope to put out a 2.7.1 release next weekend to address this bug, an anything else which might come up.

dkowis, 2021-03-30:
I don't know if helps any, but it does work with 2.6.0 for me.
[image]

JeffB42, 2021-03-30:
Thanks @dkowis, yes it helps to know that it works for some.  I have two systems/machines that I regularly test & validate against.  On one system, I was not able to reproduce this bug.  On the other, I'm able to reproduce it some of the time, but not 100%.  I think it's it a timing issue, and OctoScreen is starting before OctoPrint has had a chance to start up and finish connecting to the printer.  I added a wait of 2 seconds and that seems to have fixed the issue, and I haven't see it since.  I will be publishing 2.7.1 this upcoming weekend to fix this issue.

JeffB42, 2021-04-03:
@NicoIT31 and @dkowis - I just released v2.7.1.

closed 2021-04-03 by JeffB42