Z-Bolt/OctoScreen #285: How to turn on a pointer

grzebcio21, 2021-04-03:
Hi there, I have a 5 inch display but touch does not work. I would like to use UI with mouse but pointer is invisible. Its hard to use it without it. Any sugestions how should i turn it on?

JeffB42, 2021-04-03:
Normally one would edit `/lib/systemd/system/octoscreen.service` and remove the `-nocursor`, however there's a bug in GTK when running without a window manager, and this will require a code change and a new env variable.  I'll probably work on this this week and push an update next weekend.

JeffB42, 2021-04-10:
I just released 2.7.2.  If you install this version, and follow the steps listed at https://github.com/Z-Bolt/OctoScreen#optional-configuration-settings, the mouse's cursor will be displayed.

closed 2021-04-10 by JeffB42