Z-Bolt/OctoScreen #287: missing temperatur display

Aschinski, 2021-04-06:
Good day.

The temperature display of the hotend and heating bed are missing in the filament panel and temperature panel in the middle.

what can I do?

JeffB42, 2021-04-08:
I haven't been able to reproduce this yet, which is going to make it hard to troubleshoot, so I'm going to ask a few questions:
If you reboot your RPi (and OctoPrint and OctoScreen), does it still happen?
Is this repeatable?  Are you seeing it every time you run OctoScreen, or is it intermittent?
When you first run OctoScreen, is your printer on?  If not, try turning your printer on and then rebooting.
Have you turned logging on?  What's in your log file?  Any errors?

Aschinski, 2021-04-08:
I have reinstalled the raspberry serveral times with Octoprint and Octoscreen. However, the temperature displays were shown very rarely. If they where then displayed, they disappeared immediately after a reboot or restart. The printer was always switched on, the virtual printer was also tried, and the same problem occurred. The OctoScreen plugin was also installed.

The Raspberry was upgraded during the reinstallation.
The the dependencies were installed.
Finally, OctoScreen was installed via the .dep package.

Otherwise everything works fine.

 A Raspberry Pi 4 with 4 GB RAM is used
Octopi Version 0.18.0
Octoprint Version 1.5.3
OctoScreen Version 2.7.1
Printer: Anycubic i3 Mega S (Standard software)

I'll try to upload the log file later.

JeffB42, 2021-04-09:
Thanks, yea, please post the log file.  Be sure to set the level to debug, and when you post it, either paste it or attach it as a plain text file (.txt) and not a .DOC or .PDF.

Aschinski, 2021-04-10:
Hello! This is the log file. 
i have just reinstalled the raspberry with octopi, octoprint and octoscreen.

[octoprint (1).log](https://github.com/Z-Bolt/OctoScreen/files/6289762/octoprint.1.log)

JeffB42, 2021-04-10:
That's OctoPrint's log file, not OctoScreen.

https://github.com/Z-Bolt/OctoScreen/wiki/FAQ#q-how-do-i-enable-logging--why-would-i-want-to-turn-logging-on

Please run again and post OctoScreen's log file.

JeffB42, 2021-04-10:
Also, I just release 2.7.2, which might fix this issue.

Aschinski, 2021-04-13:
The problem is solved. i have installed octoscreen version 2.7.2 and since then no more problems! Thanks!

closed 2021-04-13 by Aschinski