Z-Bolt/OctoScreen #288: Button icons not showing

CobaltEcho, 2021-04-07:
None of the icons for the buttons are showing up. The text is there, but it's just an placeholder file icon where all the various icons are supposed to be. RPi4, latest Octoprint and Octoscreen. I can take a picture if needed.

JeffB42, 2021-04-08:
Sounds like you have a bad install.  When you installed, did you follow the install instructions (listed on the project's page, at https://github.com/Z-Bolt/OctoScreen), in the same order that's listed?
I suggest removing OctoScreen and then reinstalling.  OctoScreen loads all of its images from one location (https://github.com/Z-Bolt/OctoScreen/tree/master/styles/z-bolt/images).  I've never seen this problem before, so it sounds like you somehow have a bad installation.
Also...
* Which model of Raspberry Pi are you running?
* Which OctoPi distro/image did you install?
* Which version of OctoPrint are you running?
* Which version of OctoScreen are you running?

CobaltEcho, 2021-04-10:
Raspberry Pi 4 Model B Rev 1.4
OctoPi Version 0.18.0
OctoPrint Version 1.5.3 from https://octoprint.org/download/
OctoScreen 2.7.1

JeffB42, 2021-04-10:
> Is there a way I can uninstall and then I can try to do the installation again?

Yes, ssh into your raspberry pi, then run:
`sudo systemctl stop octoscreen.service` to stop the OctoScreen service
`sudo dpkg -r octoscreen` to remove/uninstall OctoScreen
`sudo dpkg -i octoscreen_2.7.2_armhf.deb` to install
(note: I just released 2.7.2)

JeffB42, 2021-04-10:
Also, enable the log file, reboot, and then look at the log file for errors.
https://github.com/Z-Bolt/OctoScreen/wiki/FAQ#q-how-do-i-enable-logging--why-would-i-want-to-turn-logging-on

CobaltEcho, 2021-04-10:
`time="2021-04-10T18:54:49+01:00" level=debug msg=- file=logger.go func=logger.Debug line=137
time="2021-04-10T18:54:49+01:00" level=debug msg=- file=logger.go func=logger.Debug line=137
time="2021-04-10T18:54:49+01:00" level=debug msg="entering OctoScreen - main.init()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-10T18:54:49+01:00" level=debug msg="    entering main.readConfig()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-10T18:54:49+01:00" level=info msg="        Path to OctoPrint's config file: \"/home/pi/.octoprint/config.yaml\"" file=logger.go func=logger.Infof line=150
time="2021-04-10T18:54:49+01:00" level=info msg="        main.readConfig() - server host is: \"localhost\"" file=logger.go func=logger.Infof line=150
time="2021-04-10T18:54:49+01:00" level=info msg="        main.readConfig() - server port is: 5000" file=logger.go func=logger.Infof line=150
time="2021-04-10T18:54:49+01:00" level=debug msg="    leaving main.readConfig()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-10T18:54:49+01:00" level=debug msg="    entering main.setApiKey()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-10T18:54:49+01:00" level=debug msg="        main.setApiKey() - APIKey is \"81E---374\"" file=logger.go func=logger.Debugf line=141
time="2021-04-10T18:54:49+01:00" level=debug msg="    leaving main.setApiKey()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-10T18:54:49+01:00" level=info msg="    main.setBaseUrl() - using \"http://192.168.1.120\" as server address" file=logger.go func=logger.Infof line=150
time="2021-04-10T18:54:49+01:00" level=debug msg="leaving OctoScreen - main.init()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-10T18:54:49+01:00" level=debug msg=- file=logger.go func=logger.Debug line=137
time="2021-04-10T18:54:49+01:00" level=debug msg=- file=logger.go func=logger.Debug line=137
time="2021-04-10T18:54:49+01:00" level=debug msg=+ file=logger.go func=logger.Debug line=137
time="2021-04-10T18:54:49+01:00" level=debug msg=+ file=logger.go func=logger.Debug line=137
time="2021-04-10T18:54:49+01:00" level=debug msg="entering OctoScreen - main.main()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-10T18:54:50+01:00" level=info msg="    Environment variables..." file=logger.go func=logger.Info line=146
time="2021-04-10T18:54:50+01:00" level=info msg="    Required environment variables:" file=logger.go func=logger.Info line=146
time="2021-04-10T18:54:50+01:00" level=info msg="    key: \"OCTOPRINT_HOST\", value: \"http://192.168.1.120\"" file=logger.go func=logger.Infof line=150
time="2021-04-10T18:54:50+01:00" level=info msg="    key: \"OCTOPRINT_APIKEY\", value: \"81E---374\"" file=logger.go func=logger.Infof line=150
time="2021-04-10T18:54:50+01:00" level=info msg="    key: \"OCTOSCREEN_STYLE_PATH\", value: \"/opt/octoscreen/styles/z-bolt/\"" file=logger.go func=logger.Infof line=150
time="2021-04-10T18:54:50+01:00" level=info msg="    " file=logger.go func=logger.Info line=146
time="2021-04-10T18:54:50+01:00" level=info msg="    Optional environment variables:" file=logger.go func=logger.Info line=146
time="2021-04-10T18:54:50+01:00" level=info msg="    key: \"OCTOPRINT_CONFIG_FILE\", value: \"/home/pi/.octoprint/config.yaml\"" file=logger.go func=logger.Infof line=150
time="2021-04-10T18:54:50+01:00" level=info msg="    key: \"OCTOSCREEN_LOG_FILE_PATH\", value: \"/home/pi/logs/logfile.txt\"" file=logger.go func=logger.Infof line=150
time="2021-04-10T18:54:50+01:00" level=info msg="    key: \"OCTOSCREEN_LOG_LEVEL\", value: \"debug\"" file=logger.go func=logger.Infof line=150
time="2021-04-10T18:54:50+01:00" level=info msg="    key: \"OCTOSCREEN_RESOLUTION\", value: \"800x480\"" file=logger.go func=logger.Infof line=150
time="2021-04-10T18:54:50+01:00" level=info msg="    key: \"DISPLAY_CURSOR\", value: \">>MISSING<<\"" file=logger.go func=logger.Infof line=150
time="2021-04-10T18:54:50+01:00" level=debug msg="    entering main.getSize()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-10T18:54:50+01:00" level=debug msg="    leaving main.getSize()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-10T18:54:50+01:00" level=debug msg="    entering ui.New()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-10T18:54:50+01:00" level=debug msg="        entering SplashPanel.initialize()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-10T18:54:50+01:00" level=debug msg="            entering SplashPanel.createActionBar()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-10T18:54:50+01:00" level=debug msg="            leaving SplashPanel.createActionBar()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-10T18:54:50+01:00" level=debug msg="        leaving SplashPanel.initialize()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-10T18:54:50+01:00" level=debug msg="        entering ui.initialize()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-10T18:54:50+01:00" level=debug msg="            entering ui.loadStyle()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-10T18:54:50+01:00" level=debug msg="            leaving ui.loadStyle()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-10T18:54:50+01:00" level=debug msg="            entering ui.sdNotify()" file=logger.go func=logger.TraceEnter line=101
[... cut ...]
time="2021-04-10T18:54:51+01:00" level=debug msg="                JSON response: {\"message\":\"\"}\n" file=logger.go func=logger.Debugf line=141
time="2021-04-10T18:54:51+01:00" level=debug msg="            leaving Client.doJsonRequest()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-10T18:54:51+01:00" level=debug msg="        leaving ui.checkNotification()" file=logger.go func=logger.TraceLeave line=115
time="2021-04-10T18:54:51+01:00" level=debug msg="    leaving ui.update()" file=logger.go func=logger.TraceLeave line=115
`

CobaltEcho, 2021-04-10:
> Yes, ssh into your raspberry pi, then run:
> `sudo systemctl stop octoscreen.service` to stop the OctoScreen service
> `sudo dpkg -r octoscreen` to remove/uninstall OctoScreen
> `sudo dpkg -i [...]

I uninstalled and reinstalled, same thing.  I uploaded the logfile (debug mode) in the previous comment.

JeffB42, 2021-04-10:
Hmm... I still don't have an answer yet as to what the issue is.
Just to double-check, it's all button images, and not just the ones in the System Panel?
(I'm guessing that's the case, but the attached log file was only for the Splash panel and then the Idle pane, and no errors were listed, so I just want to double-check)

In the log file, there's this:
`time="2021-04-10T18:54:50+01:00" level=info  msg="    key: \"OCTOSCREEN_STYLE_PATH\", value: \"/opt/octoscreen/styles/z-bolt/\"" file=logger.go func=logger.Infof line=150`
...which looks correct (this is what mine is set to).  This is where the images are located at.

Do the folders `/opt/octoscreen/styles/z-bolt/` and `/opt/octoscreen/styles/z-bolt/images` exist?
On my machine, when I run `ls -la` I get this:
[image]
When  you run ls -la, do you see the same files?
Do the image files have the same permissions, like `-rw-r--r--` and `root root`?

If they are the same, try this - try stopping the service and then restarting it, using sudo:
`sudo systemctl stop octoscreen.service`
`sudo systemctl start octoscreen.service`

CobaltEcho, 2021-04-10:
It seems to be only the pictures in the "images" folder that aren't showing, from what I can tell.  The "logos" sub-folder images all seem to be showing up fine.

[image]
Looks like all the image files are there (using my FTP browser).

It looks like my total counts are higher?
[image]

Also, I have the 8gb pi (64bit), but using the 32bit OctoPi (64 isnt stable yet).  Not sure if that makes a difference.

Starting and stopping the service didnt seem to make a difference.  Plugin Manager didnt make a difference either.

JeffB42, 2021-04-11:
> It looks like my total counts are higher?

Yea, that looks suspicious, but I searched and it seems the "total" that's displayed isn't the file count, it's the disk space/block size that's used.  Good ol' _intuitive_ Linux.
https://www.linuxquestions.org/questions/linux-general-1/what-total-means-in-ls-output-455323-print/#:~:text=total%20shows%20the%20amount%20of,example%3A%20ls%20%2DlH).

> It seems to be only the pictures in the "images" folder that aren't showing, from what I can tell. The "logos" sub-folder images all seem to be showing up fine.

This still seems to me to be a system setup issue.  A few posts back, I asked
_"Sounds like you have a bad install. When you installed, did you follow the install instructions (listed on the project's page, at https://github.com/Z-Bolt/OctoScreen), in the same order that's listed?"_
...but didn't get a reply back.  Did you follow the steps on the main page?

Specifically, did you run?
`sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils`

The images in /images/logos are PNG files.  The images in /images are SVGs.  Try installing librsvg (https://gitlab.gnome.org/GNOME/librsvg) though, SVG image support should have been installed with you installed those other libraries.

Install librsvg, reboot, and let me know what happens.  If the button images still don't appear, please do the following:
1. stop the octoscreen service 
2. delete the OctoScreen log file
3. re-create the OctoScreen log file (basically, I just want you to clear out your log file)
4. reboot
5. When OctoScreen starts, navigate from the Splash panel -> Idle panel -> Configuration panel -> System panel
6. At each panel, take a photo and then post them.  Or record a video of the process - I think GitHub now supports posting videos (or upload the video to YouTube).
7. Also post the new log file.

CobaltEcho, 2021-04-11:
I did follow the instructions on the home page, including the server dependencies. 

Log:  [logfile.txt](https://github.com/Z-Bolt/OctoScreen/files/6292273/logfile.txt)
Video: https://drive.google.com/file/d/1Xbmw3vbjq9nhb3YA1SYdA2DQFJq35Tje/view?usp=sharing

JeffB42, 2021-04-11:
Thanks, but when I try to view the video google is asks for access.  Could you change the video to being public?

CobaltEcho, 2021-04-11:
> Thanks, but when I try to view the video google is asks for access. Could you change the video to have public access?
Oops, this should work:
https://drive.google.com/file/d/1Xbmw3vbjq9nhb3YA1SYdA2DQFJq35Tje/view?usp=sharing

JeffB42, 2021-04-11:
"Makes a nice mirror"  lol

Ok, this helps.  Looking at log file, I see some errors, like:
```
time="2021-04-11T17:41:18+01:00" level=error msg="                gtk.MustImageFromFileWithSize() - PixbufNewFromFileAtScale() returned an error: \"Couldn’t recognise the image file format for file “/opt/octoscreen/styles/z-bolt/images/back.svg”\"" file=logger.go func=logger.LogError line=121
```

and...
```
time="2021-04-11T17:41:53+01:00" level=debug msg="    entering networkPanel.update()" file=logger.go func=logger.TraceEnter line=101
time="2021-04-11T17:41:53+01:00" level=error msg="        gtk.MustImageFromFileWithSize() - PixbufNewFromFileAtScale() returned an error: \"Couldn’t recognise the image file format for file “/opt/octoscreen/styles/z-bolt/images/network.svg”\"" file=logger.go func=logger.LogError line=121
time="2021-04-11T17:41:53+01:00" level=error msg="        gtk.MustImageFromFileWithSize() - PixbufNewFromFileAtScale() returned an error: \"Couldn’t recognise the image file format for file “/opt/octoscreen/styles/z-bolt/images/network.svg”\"" file=logger.go func=logger.LogError line=121
time="2021-04-11T17:41:53+01:00" level=error msg="        gtk.MustImageFromFileWithSize() - PixbufNewFromFileAtScale() returned an error: \"Couldn’t recognise the image file format for file “/opt/octoscreen/styles/z-bolt/images/network.svg”\"" file=logger.go func=logger.LogError line=121
time="2021-04-11T17:41:53+01:00" level=error msg="        gtk.MustImageFromFileWithSize() - PixbufNewFromFileAtScale() returned an error: \"Couldn’t recognise the image file format for file “/opt/octoscreen/styles/z-bolt/images/network.svg”\"" file=logger.go func=logger.LogError line=121
time="2021-04-11T17:41:53+01:00" level=error msg="        gtk.MustImageFromFileWithSize() - PixbufNewFromFileAtScale() returned an error: \"Couldn’t recognise the image file format for file “/opt/octoscreen/styles/z-bolt/images/network.svg”\"" file=logger.go func=logger.LogError line=121
time="2021-04-11T17:41:53+01:00" level=error msg="        gtk.MustImageFromFileWithSize() - PixbufNewFromFileAtScale() returned an error: \"Couldn’t recognise the image file format for file “/opt/octoscreen/styles/z-bolt/images/network.svg”\"" file=logger.go func=logger.LogError line=121
time="2021-04-11T17:41:53+01:00" level=debug msg="    leaving networkPanel.update()" file=logger.go func=logger.TraceLeave line=115
```

However, this is only a few of the (SVG) images...  in the video (thanks for posting btw!) it appears that every PNG is displayed, but none of the SVGs are.

Try installing librsvg and then rebooting.

Also, in your video, that first button that you pressed (upper left hand corner)... that's the button which turns on/turns off the hotend, so be careful with that.  If you accidentally press it (or the heat bed button), tapping it again will turn it of.

JeffB42, 2021-04-11:
Also, run:
``apt list --installed`
...and look for librsvg.  On my machine it's:
```
librsvg2-2/stable,now 2.44.10-2.1+rpi1 armhf [installed,automatic]
librsvg2-common/stable,now 2.44.10-2.1+rpi1 armhf [installed,automatic]
```

CobaltEcho, 2021-04-11:
When I run
`pi@octopi:/ $ ldconfig -p | grep librsvg`

I get:
`librsvg-2.so.2 (libc6,AArch64) => /usr/lib/aarch64-linux-gnu/librsvg-2.s                        o.2`

`apt list --installed`
Gives me:
`librsvg2-2/stable,now 2.44.10-2.1 arm64 [installed,automatic]`
`librsvg2-bin/stable,now 2.44.10-2.1 arm64 [installed]`
`librsvg2-common/stable,now 2.44.10-2.1 arm64 [installed,automatic]`

JeffB42, 2021-04-11:
I'm running out of suggestions to try.  The results I cited before were on a RPi 3B.  I just switched over to my RPi4 (only 2GB) and ran the same commands, and got:
```
$ ldconfig -p | grep librsvg
    librsvg-2.so.2 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/librsvg-2.so.2

$ apt list --installed
    librsvg2-2/stable,now 2.44.10-2.1+rpi1 armhf [installed,automatic]
    librsvg2-common/stable,now 2.44.10-2.1+rpi1 armhf [installed,automatic]
```
...so my RPi3 install is nearly the same as my RPi4 install, only that librsvg-2.so.2 in in /lib on one machine and in /usr/lib on another, but other than that, my two librsvg are the same.  Yet they differ from what you have.

The version of OctoPrint that I'm running is the same as yours (1.5.3) but my OctoPi version is 0.17.0.

OK, try this:
$ sudo apt-get remove librsvg
$ sudo apt-get install librsvg

(I didn't run this, but I think those are the right commands)

If this STILL doesn't work, do you have a spare SD card?  Try flashing OctoPi 0.17.0 onto the spare SD card, configure OctoPrint, and then install OctoScreen and run through the steps listed on the main page.

Jeff

JeffB42, 2021-04-11:
One other thing - you said earlier:
>Also, I have the 8gb pi (64bit), but using the 32bit OctoPi (64 isnt stable yet). Not sure if that makes a difference.

From what you listing in apt list, is sounds like you are using the 64 bit version.
You have
`librsvg2-2/stable,now 2.44.10-2.1 arm64`
...but I have:
`librsvg2-2/stable,now 2.44.10-2.1+rpi1 armhf`

armf is for 32-but architecture, but arm64 is for 64 bit.
Double check this please.  You said you are using 32-bit, yet the binaries on your machine are arm64.

CobaltEcho, 2021-04-11:
Crap, you're right. I tried the 64bit nightly and didn't get any to work.
Tried the 32bit without luck. Eventually found a 64bit nightly Octopi that
worked.

So yeah, I have 64bit.  :/

JeffB42, 2021-04-17:
> I do find it odd that the svg logo pictures are fine, and it's only the icons

The images that are appearing are PNGs.
The images which aren't appearing are SVGs.

> I'm guessing OctoScreen doesn't support 64bit yet?

I haven't tried.  It's a 32-but app.  I think the good news is that it appears that the app mostly works, it's only the SVGs that aren't appearing.  I suggest either:
a) removing the 64-bit version of librsvg and then installing the 32-bit version.

https://www.google.com/search?q=linus+install+32+but+version+of+library+on+64+bit+linux&oq=linus+install+32+but+version+of+library+on+64+bit+linux&aqs=chrome..69i57j33i299.14687j0j7&sourceid=chrome&ie=UTF-8

I have no idea if this will work.  Good luck.

b) start over, and use a 32-bit version of OctoPrint (official version, not nightly build)
https://octoprint.org/download/

> I originally tried the 64bit nightly and didn't get any to work. Then tried the 32bit without luck. Eventually found a 64bit nightly Octopi that worked.

I don't know what's going on, but I've never had any issues with their official builds.  Honestly, my best advice (and this has always worked for me, w/o any issues at all) is to go to
https://octoprint.org/download/
download the official image (not nightly build)
flash SD card with the image
boot and the then run through the instructions listed at https://octoprint.org/download/

closed 2021-04-17 by JeffB42

CobaltEcho, 2021-04-17:
> The images that are appearing are PNGs.
> The images which aren't appearing are SVGs.
>
>
> I haven't tried. It's a 32-but app. I think the good news is that it appears that the app mostly works, it's [...]

> [...

The official 32bit builds work great on 32bit boards. There is no official 64bit (yet), only nightly builds.