Z-Bolt/OctoScreen #292: PANIC on start up if a custom system action defined Octoprint does not contain a 'confirm' attribute

freddieflintstone, 2021-04-14:
Like the title says OctoScreen fails to load if a "confirm" attribute is not defined for Octoprint's custom system action, here are the errors...

```
... SystemCommandsRequest.Do() - json.Unmarshal(Custom) returned an error: \"unexpected end of JSON input\"" file=logger.go func=logger.LogError line=121
... control.getCommands() - Do(SystemCommandsRequest) returned an error: \"unexpected end of JSON input\"" file=logger.go func=logger.LogError line=121
... SystemCommandsRequest.Do() - json.Unmarshal(Custom) returned an error: \"unexpected end of JSON input\"" file=logger.go func=logger.LogError line=121
... PANIC!!! - CreateSystemCommandButton() - SystemCommandsRequest.Do() returned an error: \"unexpected end of JSON input\"" file=logger.go func=logger.LogError line=121
```

Although I didn't want a confirmation dialog on the particular action causing the issue above, I went ahead and added one to work-around the problem. As a 'confirm' attribute is not required on a system action in Octoprint, then Octoscreen should probably follow the same and not PANIC if it doesn't exist. Just my 2 cents. 

Keep up the good work!

JeffB42, 2021-04-17:
Hmm... I haven't been able to reproduce this issue.  Both scripts which require a conformation and script which don't work. 
 I'm using the latest release of OctoPrint, and I'm also using Custom Control Editor plug-in to create the scripts.  To try & repro this, I created:
a command (with 'Use Confirmation' off)
a command (with 'Use Confirmation' on)
a script command (with 'Use Confirmation' off)
a script command (with 'Use Confirmation' on)

And all four are working.

Are you using the Custom Control Editor plug-in?
If you are, could you post a screen shot of the command that causes the panic?

If you aren't using this plug-in, how are you creating the command?  What's the content of the command?

freddieflintstone, 2021-04-17:
I didn't use the plugin to add the system action, I had edited octoprint's config.xml directly. Here's what the custom actions looked like which causes the octoscreen panic...

```
system:
    actions:
    -   action: printer_on
        command: gpio -g write 21 1
        name: Turn printer on
    -   action: printer_off
        command: gpio -g write 21 0
        confirm: You are about to turn off the printer
        name: Turn printer off
```

JeffB42, 2021-04-25:
@freddieflintstone Thanks for reporting this, I just release v2.7.3, and it fixes this issue.

closed 2021-04-25 by JeffB42