Z-Bolt/OctoScreen #293: Cannot get screen correct size on PiTFT 3.5

tjscientist, 2021-04-15:
I have spent hours configuring, re-configuring and researching how to get this display to work correctly. It was working fine until I decided to upgrade Octoprint and octoscreen to the latest versions so I could have built in Python 3 (for the bed mesh visualization). Now, even with the same octoscreen/config and config.txt files I used previously, the content is about double the size of the display.

[image]

I have tried starting from a fresh installation, using the Adafruit installation scripts, manually installing and nothing seems to work. I have followed many of the online tutorials and they don't work.

Either I have something really simply I am overlooking (most likely the case) or something changes in the latest updates. Any help would be greatly appreciated.

tjscientist, 2021-04-16:
Started again from blank SD.
Installed latest OctoPi Image
Did update / upgrade
rebooted
updated to latest OctoPrint
Installed PiTFT (mirror HDMI)
edited touch display in X11 (touchmatrix)
installed OctoScreen
rebooted
Added plugins (bed level visualizer, etc.)
rebooted
everything seems to work. Not sure what I was doing wrong... Maybe I just needed a good night sleep.

closed 2021-04-16 by tjscientist

JeffB42, 2021-04-17:
Glad you got it configured and working.

> Maybe I just needed a good night sleep.

That helps.  Plus, the setup steps need to be run, and in the order listed.  It can some time be easy to miss this.