Z-Bolt/OctoScreen #295: Prusa Thumbnails

TMProjection, 2021-04-23:
Hey all, I have Prusa MK3S+, with OctoPi on Raspberry 4 Type B

I have Prusa Thumbnails plugin installed and it displays the thumbnail on OctoScreen when i tap on "print file" however during the print itself it doesnt shot the thumbnail like OctoDash can do

Is there anyway to enable them?

JeffB42, 2021-04-25:
The Print panel doesn't display a thumbnail, only the file select panel does.  Also, in the Print panel, there isn't any space (screen real estate) to draw the thumbnail.  There's another ticket that's open, requesting thew thumbnail be drawn when printing, please see my reply there, https://github.com/Z-Bolt/OctoScreen/issues/283

closed 2021-04-25 by JeffB42

TMProjection, 2021-04-25:
Honestly i would have to disagree, as a designer myself, visually theres plenty of room to rearrange, programming wise im not sure but graphically i would be happy to help out if i can?