Z-Bolt/OctoScreen #298: Octoscreen 2.7.3

CountryDevil, 2021-05-01:
After setting up a new device and testing, I noticed that "back arrow" on all screens on this version does not seem to work.  Touching it does nothing. All other "buttons" seem to be working.

JeffB42, 2021-05-01:
I haven't been able to reproduce the issue.  I booted up 2.7.3 on my RPi 3B, and after it connected to the printer and Idle screen appeared...
I pressed the Home button, and then then pressed the Back button
I pressed the Actions button, and then then pressed the Back button 
I pressed the Filament button, and then then pressed the Back button 
...and everything worked as expected.

I suppose the first thing to suggest (ala The IT Crowd) is try turning it off and on again.  Not just rebooting, but powering off the display and CPU.

Assuming that doesn't fix the issue, if what is displayed appears correct, but the buttons aren't responding like they should, it sounds like you have a configuration issue, and the display (pixels) is set up correctly, but the touch input is not.  Possibly the wrong resolution, possible it needs to be rotated.

CountryDevil, 2021-05-02:
After a bit more troubleshooting, this issue appears to be hardware related on a faulty Raspberry Pi.  I was able to get a new Pi4 yesterday and after the swap, things seems to be working fine now. 

Sorry for any confusion this may have caused. Troubleshooting 101 stuff here and I failed myself.

JeffB42, 2021-05-03:
No worries, glad it's working for you.

closed 2021-05-03 by JeffB42