Z-Bolt/OctoScreen #299: Unable to cancel Filament Loading / Unloading once started

mattydboom, 2021-05-26:
When I use the Filament loading button, on my Qidi Tech 1 (Replicator 1 clone - dual head) printer, then the **filament is extruded continuously (as I guess it is intended) BUT I am unable to stop it**. Pushing the Load button again doesn't work, and neither does the Unload or the Back button etc... all have no effect. So I have to quickly use the printers own LCD screen & buttons to navigate to printer loading menus and stop it.

Is this a known issue, or perhaps something specific to the Marlin / Sailfish perhaps (mine is running version 7.8). I notice for other OctoScreen buttons such as preheating the Nozzles etc. that the button has a grey background applied to it when toggled 'ON'.... so I would have expected the same to have been the case for these Filament Load / unload buttons, but once pushed the background is unchanged, even though the code is clearly being sent.

mattydboom, 2021-05-26:
Ok so I may have solved this by reading through old posts. In particular this one; https://github.com/Z-Bolt/OctoScreen/issues/100

Delving into the OctoScreen Plugin setting I see now that the load/unload duration are determined by the filament Change length parameters. By default these were much too high for my machine (750 & 800mm) which gave the impression that the load/unload command was never going to stop. Having reduced these to more suitable values I am back up and running. 

**As a suggestion, perhaps these values could appear on or beside the load / unload icons? This would be a useful prompt before hitting the button and so you can make a mental note to change the plugin settings if insufficient. (Much like the Extrude/Retract buttons have the 3-stage button which states the length that the extruder will move per press)**

I must also take this opportunity to say how extremely good this software is. I researched heavily before installing, and really think its the best choice available, so thank you developers for all the hard work. I look forward to future updates :)

[image]

JeffB42, 2021-06-02:
This is something that's been a little tricky.  Originally, the in/out length was REALLY long (750mm, if I'm remembering correctly).  This was set for machines with Bowden extruders, but this wreaked havoc on machines with a direct drive, so I compromised and set it to 100mm.  As you've discovered, 100mm in/out is the default and one can change their values, but the plug-in needs to be installed.

closed 2021-06-02 by JeffB42