Z-Bolt/OctoScreen #301: OctoScreen restarts every ~6 minutes due to Watchdog timeout

AndyDiags, 2021-05-30:
While printing, my OctoScreen restarts every ~6 minutes.

The time elapsed/remaining slows down, and eventually becomes unresponsive and reboots
Always occurring, even after reboot.

System logs show reboot occurs due to Watchdog timeout.

System Info:
OctoPrint v1.6.1
OctoScreen 2.7.1
OctoScreen plugin 2.6.0
Waveshare 4.3inch DSI LCD

Resoluton: 800x480
Raspberry Pi 4 4GB
Prusa MK3S

Additional Info:
-RPi is sufficiently powered
-Only OctoScreen is reseting, OctopPrint logs show no indication of any errors
-If i stay on the "System Information" screen or any other screen, the OctoScreen does not crash

```
May 31 00:36:26 octopi systemd[1]: octoscreen.service: Watchdog timeout (limit 20s)!
May 31 00:36:26 octopi systemd[1]: octoscreen.service: Killing process 4119 (xinit) with signal SIGABRT.
May 31 00:36:26 octopi systemd[1]: octoscreen.service: Killing process 4120 (Xorg) with signal SIGABRT.
May 31 00:36:26 octopi xinit[4119]: (EE)
May 31 00:36:26 octopi xinit[4119]: (EE) Backtrace:
May 31 00:36:26 octopi xinit[4119]: (EE)
May 31 00:36:26 octopi xinit[4119]: (EE) Received signal 6 sent by process 1, uid 0
May 31 00:36:26 octopi xinit[4119]: (EE)
May 31 00:36:26 octopi xinit[4119]: Fatal server error:
May 31 00:36:26 octopi xinit[4119]: (EE) Caught signal 6 (Aborted). Server aborting
May 31 00:36:26 octopi xinit[4119]: (EE)
May 31 00:36:26 octopi xinit[4119]: (EE)
May 31 00:36:26 octopi xinit[4119]: Please consult the The X.Org Foundation support
May 31 00:36:26 octopi xinit[4119]: #011 at http://wiki.x.org
May 31 00:36:26 octopi xinit[4119]:  for help.
May 31 00:36:26 octopi xinit[4119]: (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
May 31 00:36:26 octopi xinit[4119]: (EE)
May 31 00:36:26 octopi xinit[4119]: (II) AIGLX: Suspending AIGLX clients for VT switch
May 31 00:36:26 octopi systemd[1]: octoscreen.service: Killing process 4150 (OctoScreen) with signal SIGABRT.
May 31 00:36:26 octopi systemd[1]: octoscreen.service: Main process exited, code=killed, status=6/ABRT
May 31 00:36:26 octopi xinit[4119]: SIGABRT: abort(EE) Server terminated with error (1). Closing log file.
May 31 00:36:26 octopi systemd[1]: octoscreen.service: Failed with result 'watchdog'.
May 31 00:36:26 octopi systemd[1]: octoscreen.service: Service RestartSec=100ms expired, scheduling restart.
May 31 00:36:26 octopi systemd[1]: octoscreen.service: Scheduled restart job, restart counter is at 99.
May 31 00:36:26 octopi systemd[1]: Stopped OctoScreen.
May 31 00:36:26 octopi systemd[1]: Starting OctoScreen...
May 31 00:36:27 octopi xinit[4775]: X.Org X Server 1.20.4
May 31 00:36:27 octopi xinit[4775]: X Protocol Version 11, Revision 0
May 31 00:36:27 octopi xinit[4775]: Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
May 31 00:36:27 octopi xinit[4775]: Current Operating System: Linux octopi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l
May 31 00:36:27 octopi xinit[4775]: Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:89:10:24 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=bd8064d1-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait spidev.bufsiz=32768
```

mattydboom, 2021-06-01:
Mine does this too, and didn't know if it was intentional or a glitch.

JeffB42, 2021-06-02:
I have not seen this before.  Not while printing, anyway.  @AndyDiags and @mattydboom does the OctoScreen reset happen when idle?

> If i stay on the "System Information" screen or any other screen, the OctoScreen does not crash

OK, so only when printing then, right?

@AndyDiags thanks for uploading the details of your system, and the video - it helps.  You uploaded the system's log file though.  Could you turn on logging and set the level to debug, and then post a plan-text file of the log?  Instructions for doing this are in the Wiki, at https://github.com/Z-Bolt/OctoScreen/wiki/FAQ#q-how-do-i-enable-logging--why-would-i-want-to-turn-logging-on.

@AndyDiags also, two other things:
1. Update to OctoScreen 2.7.3.  2.7.2 and 2.7.3 addressed some crashing issues, and also has some additional logging.
2. Other than OctoPrint and OctoScreen (and excluding OctoPrint plugins), did you install any other hardware or software on your RPi4?  I have a Prusa MK3S as well, and have used OctoScreen with a RPi4 and a RPi3B, and not had this issue.

mattydboom, 2021-06-02:
@ JeffB42 no I have only noticed this behaviour when a print is running. Thankfully the prints themselves continue to run unaffected during the restart....so I thought it may be normal behaviour but good to see this logged as a bug and hopefully something we can help to identify the route cause and get fixed. I will upload a log fie as per your instructions above on my next print later today.

In the mean time for you info - I am running;

Qidi Tech 1 (Replicator 1 clone - dual head) printer, Marlin / Sailfish (mine is running v7.8)
OctoScreen (v2.7.3), Z-Bolt OctoScreen (2.6.0)
OctoPrint (0.18.0) on a Raspberry Pi 4B (4GB) paired with a Bigtreetech PI TFT50 Touch screen display

Apart from a Logitec Lifecam HD3000 I have no other hardware /peripherals installed.

mattydboom, 2021-06-02:
@JeffB42 I have changed the config file settings as per the instructions, but having trouble when exiting the nano & saving the config file..... which option should I be selecting do you know?
 
[image]

AndyDiags, 2021-06-04:
> @JeffB42 I have changed the config file settings as per the instructions, but having trouble when exiting the nano & saving the config file..... which option should I be selecting do you know?
> 
> [image]

Hi Matt,
When you get to the screen seen in the screenshot, all you need to do is press enter again.

To exit nano, hold down the Ctrl key and press the x key (a combination we call ctrl + x in this book).  If you have created or altered some text but have not yet saved it, nano asks:

First initiate file close by pressing Ctrl key and press the x.  Next, to save the changes, just type y and you will be prompted for a destination filepath. (as seen in screenshot).  Lastly, just press enter to save to the pre-filled in file path.

AndyDiags, 2021-06-04:
Did a lot more troubleshooting today...

-After updating to OctoScreen 2.7.3, I still see the issue.
-I disabled all plugins except for Octoprint and reboot, but still see the issue.
-If I disable the Camera (in the "Control" tab of Octoprint, the issue no longer occurs.

I'm still seeing this crash.

[Debug Updated Log](https://github.com/Z-Bolt/OctoScreen/files/6600469/Debug.Updated.Log)

javierzg, 2021-06-06:
Greetings and I hope you are all well.

Exactly the same thing happens to me as AndyDiags, the difference I have:
Raspberry Pi 4 B 4 Gigas
Ender 3 V2
OctoPrint v1.6.1
OctoScreen 2.7.3
OctoScreen Plugin "Not Installing" why it is not installed I do not know ?
7 "inch HDMI LCD Touch screen.

JeffB42, 2021-08-08:
I release v2.7.4 which fixes an issue where the service was restarting.  Try installing 2.7.4 and let me know if there's still an issue.  If there's still an issue, please post an updated debug log.

closed 2021-08-29 by JeffB42

AndyDiags, 2021-09-12:
I've updated to the newest version and confirm no longer seeing screen reset issue.

OctoPrint v1.6.1
OctoScreen 2.7.4
OctoScreen plugin 2.6.0
Prusa MK3S v3.10.0-4481