Z-Bolt/OctoScreen #304: How can I get an EzPlug button from OctoPrint on my OctoScreen

ecwhitney42, 2021-06-10:
I installed OctoScreen and it works great but I can't find any way to add the power control button for my EzPlug that I have on my OctoPrint screen (it's the Tasmoto plugin).

I'm hoping there's a user-configurable way to do this as I would expect that other users might have OctoPrint features they'd like to bring to OctoScreen.

ecwhitney42, 2021-08-18:
Seriously, nobody else has wanted to do this? It seems like such an obvious feature--be able to turn on your printer from the Pi screen so that OctoPrint can turn it off when the print is done?

JeffB42, 2021-08-29:
There are already a few requests for this.  https://github.com/Z-Bolt/OctoScreen/issues/31 https://github.com/Z-Bolt/OctoScreen/issues/143

ecwhitney42, 2021-08-30:
Fantastic! I'll keep my eyes out for an update. I'm willing to test a build if you want someone with the hardware setup to do it (specifically with the EzPlug). Thanks! -Erich