Z-Bolt/OctoScreen #311: Fresh Install, screen keeps restarting

Moonbase1, 2021-08-04:
Fresh install of Octoprint and Octoscreen. Last night octoscreen was running fine, I let it print when i went to bed and apparently the screen froze at one point (but the Pi was still running, could SSH into it and control it from TSD and Octoprint Browser). I restarted the Pi and every few minutes it goes from Octoscreen to the terminal and then back. 

Unrelated: Is it possible to also run TouchUI and Octoscreen so when I use my phone I can use TouchUI? or will it have a conflict? (I do not have TouchUI installed, just wondering before I do).

pi@octopi:~ $ systemctl status octoscreen.service
● octoscreen.service - OctoScreen
   Loaded: loaded (/lib/systemd/system/octoscreen.service; enabled; vendor preset: enabled)
   Active: activating (start) since Wed 2021-08-04 13:49:02 PDT; 711ms ago
 Main PID: 20468 (xinit)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/octoscreen.service
           ├─20468 /usr/bin/xinit /usr/bin/OctoScreen -- :0 -nolisten tcp -nocursor
           └─20469 /usr/lib/xorg/Xorg :0 -nolisten tcp -nocursor

Aug 04 13:49:02 octopi xinit[20468]: Build Date: 19 April 2021  03:23:36PM
Aug 04 13:49:02 octopi xinit[20468]: xorg-server 2:1.20.4-1+rpt3+deb10u3 (https://www.debian.org/support)
Aug 04 13:49:02 octopi xinit[20468]: Current version of pixman: 0.36.0
Aug 04 13:49:02 octopi xinit[20468]:         Before reporting problems, check http://wiki.x.org
Aug 04 13:49:02 octopi xinit[20468]:         to make sure that you have the latest version.
Aug 04 13:49:02 octopi xinit[20468]: Markers: (--) probed, (**) from config file, (==) default setting,
Aug 04 13:49:02 octopi xinit[20468]:         (++) from command line, (!!) notice, (II) informational,
Aug 04 13:49:02 octopi xinit[20468]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Aug 04 13:49:02 octopi xinit[20468]: (==) Log file: "/var/log/Xorg.0.log", Time: Wed Aug  4 13:49:02 2021
Aug 04 13:49:02 octopi xinit[20468]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"

JeffB42, 2021-08-05:
> I restarted the Pi and every few minutes it goes from Octoscreen to the terminal and then back.

Turn on logging and set the level to debug.  Instructions are in the FAQ. https://github.com/Z-Bolt/OctoScreen/wiki/FAQ

> Is it possible to also run TouchUI and Octoscreen

I don't think so, no.

Moonbase1, 2021-08-05:
> > I restarted the Pi and every few minutes it goes from Octoscreen to the terminal and then back.
> 
> Turn on logging and set the level to debug. Instructions are in the FAQ. https://github.com/Z-Bolt/OctoScreen/wiki/FAQ
> 
> > Is it possible to also run TouchUI and Octoscreen
> 
> I don't think so, no.

enabled logging, installed the plugin to octoprints UI also, and it's still doing this. 

It worked until it froze, and now it does this. It's not restarting the application - when it's midprint, the progress will stay and everything. It will blank out, show the terminal, then come back.

Moonbase1, 2021-08-05:
GNU nano 3.2                                                                                                     logfile.txt                                                                                                               
time="2021-08-04T22:30:23-07:00" level=debug msg=- file=logger.go func=logger.Debug line=137
time="2021-08-04T22:30:23-07:00" level=debug msg=- file=logger.go func=logger.Debug line=137
time="2021-08-04T22:30:23-07:00" level=debug msg="entering OctoScreen - main.init()" file=logger.go func=logger.TraceEnter line=101
time="2021-08-04T22:30:23-07:00" level=debug msg="    entering main.readConfig()" file=logger.go func=logger.TraceEnter line=101
time="2021-08-04T22:30:23-07:00" level=info msg="        Path to OctoPrint's config file: \"/home/pi/.octoprint/config.yaml\"" file=logger.go func=logger.Infof line=150
time="2021-08-04T22:30:23-07:00" level=info msg="        main.readConfig() - server host is: \"localhost\"" file=logger.go func=logger.Infof line=150
time="2021-08-04T22:30:23-07:00" level=info msg="        main.readConfig() - server port is: 5000" file=logger.go func=logger.Infof line=150
time="2021-08-04T22:30:23-07:00" level=debug msg="    leaving main.readConfig()" file=logger.go func=logger.TraceLeave line=115
time="2021-08-04T22:30:23-07:00" level=debug msg="    entering main.setApiKey()" file=logger.go func=logger.TraceEnter line=101
time="2021-08-04T22:30:23-07:00" level=debug msg="        main.setApiKey() - APIKey is empty, now using cfg.API.Key" file=logger.go func=logger.Debug line=137
time="2021-08-04T22:30:23-07:00" level=debug msg="        main.setApiKey() - APIKey is \"2EA---886\"" file=logger.go func=logger.Debugf line=141
time="2021-08-04T22:30:23-07:00" level=debug msg="    leaving main.setApiKey()" file=logger.go func=logger.TraceLeave line=115
time="2021-08-04T22:30:23-07:00" level=info msg="    main.setBaseUrl() - using \"http://localhost:5000\" as server address" file=logger.go func=logger.Infof line=150
time="2021-08-04T22:30:23-07:00" level=debug msg="leaving OctoScreen - main.init()" file=logger.go func=logger.TraceLeave line=115
time="2021-08-04T22:30:23-07:00" level=debug msg=- file=logger.go func=logger.Debug line=137
time="2021-08-04T22:30:23-07:00" level=debug msg=- file=logger.go func=logger.Debug line=137
time="2021-08-04T22:30:23-07:00" level=debug msg=+ file=logger.go func=logger.Debug line=137
time="2021-08-04T22:30:23-07:00" level=debug msg=+ file=logger.go func=logger.Debug line=137
time="2021-08-04T22:30:23-07:00" level=debug msg="entering OctoScreen - main.main()" file=logger.go func=logger.TraceEnter line=101
time="2021-08-04T22:30:24-07:00" level=info msg="    System Information..." file=logger.go func=logger.Info line=146
time="2021-08-04T22:30:24-07:00" level=info msg="    OctoScreen version: \"2.7.3\"" file=logger.go func=logger.Infof line=150
time="2021-08-04T22:30:24-07:00" level=info msg="    " file=logger.go func=logger.Info line=146
time="2021-08-04T22:30:24-07:00" level=info msg="    Environment variables..." file=logger.go func=logger.Info line=146
time="2021-08-04T22:30:24-07:00" level=info msg="    Required environment variables:" file=logger.go func=logger.Info line=146
time="2021-08-04T22:30:24-07:00" level=info msg="    key: \"OCTOPRINT_HOST\", value: \"http://localhost:5000\"" file=logger.go func=logger.Infof line=150
time="2021-08-04T22:30:24-07:00" level=info msg="    key: \"OCTOPRINT_APIKEY\", value: \">>MISSING<<\"" file=logger.go func=logger.Infof line=150
time="2021-08-04T22:30:24-07:00" level=info msg="    key: \"OCTOSCREEN_STYLE_PATH\", value: \"/opt/octoscreen/styles/z-bolt/\"" file=logger.go func=logger.Infof line=150
time="2021-08-04T22:30:24-07:00" level=info msg="    " file=logger.go func=logger.Info line=146
time="2021-08-04T22:30:24-07:00" level=info msg="    Optional environment variables:" file=logger.go func=logger.Info line=146
time="2021-08-04T22:30:24-07:00" level=info msg="    key: \"OCTOPRINT_CONFIG_FILE\", value: \"/home/pi/.octoprint/config.yaml\"" file=logger.go func=logger.Infof line=150
time="2021-08-04T22:30:24-07:00" level=info msg="    key: \"OCTOSCREEN_LOG_FILE_PATH\", value: \"/home/pi/logs/logfile.txt\"" file=logger.go func=logger.Infof line=150
time="2021-08-04T22:30:24-07:00" level=info msg="    key: \"OCTOSCREEN_LOG_LEVEL\", value: \"debug\"" file=logger.go func=logger.Infof line=150
time="2021-08-04T22:30:24-07:00" level=info msg="    key: \"OCTOSCREEN_RESOLUTION\", value: \"800x480\"" file=logger.go func=logger.Infof line=150
time="2021-08-04T22:30:24-07:00" level=info msg="    key: \"DISPLAY_CURSOR\", value: \">>MISSING<<\"" file=logger.go func=logger.Infof line=150
time="2021-08-04T22:30:24-07:00" level=info msg="    " file=logger.go func=logger.Info line=146
time="2021-08-04T22:30:24-07:00" level=debug msg="    entering main.getSize()" file=logger.go func=logger.TraceEnter line=101
time="2021-08-04T22:30:24-07:00" level=debug msg="    leaving main.getSize()" file=logger.go func=logger.TraceLeave line=115
time="2021-08-04T22:30:24-07:00" level=debug msg="    entering ui.New()" file=logger.go func=logger.TraceEnter line=101
time="2021-08-04T22:30:24-07:00" level=debug msg="        entering SplashPanel.initialize()" file=logger.go func=logger.TraceEnter line=101
time="2021-08-04T22:30:24-07:00" level=debug msg="            entering SplashPanel.createActionBar()" file=logger.go func=logger.TraceEnter line=101
[... cut ...]
time="2021-08-04T22:30:24-07:00" level=debug msg="        entering ui.verifyConnection()" file=logger.go func=logger.TraceEnter line=101
time="2021-08-04T22:30:24-07:00" level=debug msg="            ui.verifyConnection() - about to call ConnectionRequest.Do()" file=logger.go func=logger.Debug line=137
time="2021-08-04T22:30:24-07:00" level=debug msg="            entering ConnectionRequest.Do()" file=logger.go func=logger.TraceEnter line=101
time="2021-08-04T22:30:24-07:00" level=debug msg="                entering Client.doJsonRequest()" file=logger.go func=logger.TraceEnter line=101

JeffB42, 2021-08-08:
I released v2.7.4, which fixes this issue (I hope!)  Install 2.7.4 and let me know if you still get this issue.

mattydboom, 2021-08-08:
HI @JeffB42 just installing this update now to test. Noticed a small typ-o that prevents the ssh command from successfully downloading the newest files and gives a 404 error. Just changed the 3 to a 4 and everything ran perfectly, so thought you may want to update the release notes so others don't bump into the same issue. Thanks! 

[image]

JeffB42, 2021-08-08:
Whops!  Thanks for letting me know, I just updated it.

Moonbase1, 2021-08-10:
> I released v2.7.4, which fixes this issue (I hope!) Install 2.7.4 and let me know if you still get this issue.

It happened once as far as I saw. But it's definitely stable now. One thing I did notice when I had a failure before you realized the new update, was the pause and stop buttons weren't functioning. The readout was up to date (percentage, etc)

JeffB42, 2021-08-29:
Thanks @Moonbase1 , it should like this is fixed now, so I'm going to close this ticket.

closed 2021-08-29 by JeffB42