Z-Bolt/OctoScreen #314: Time Left freezes during long prints

Moonbase1, 2021-08-11:
I just did a 24 hour print. Eventually the time left just freezes and would only update randomly. at 98/99% It went from 31 minutes to 15 minutes left.

JeffB42, 2021-08-29:
I haven't encountered this.  1) Are you using the latest version (2.7.4)?  2) How often does this happen?  (every print, or every once in a while?)  3) The next time this happens, log into OcotPrint using your browser, and compare what's displayed in the browser window, with

AndyDiags, 2021-09-15:
I had same issue where my time left was wildly incorrect for my Prusa printer, and noticed it was also displaying incorrectly in the browser window.  To resolve, I added a plugin which allows OctoPrint to pull the time left from the gcode instead of trying to estimate:
https://github.com/gdombiak/OctoPrint-M73ETAOverride
Now the time left on my OctoScreen matches time left on my Prusa LCD.