Z-Bolt/OctoScreen #317: WIFI information screen bug

obrhaberer, 2021-08-31:
Octoscreen is running on the official raspberry pi touch screen 800 * 480. I'am using display_rotate=2 in config.
Everything is working except when I'm pressing "WIFI" the screen somehow moves the information out of place.

[image]

AndyDiags, 2021-09-15:
I see two issues here.  

One issue is that there's something wrong with your Wi-Fi.  Hard to tell as the rest of the message is cut off, maybe it says "INTERFACE IS OFF".  Would need logs to figure out, but probably not OctoScreen related.

Second issue is that the screen size is not correct on this menu.  I can somewhat reproduce this with my 800x480 screen.  In my instance, the screen gets resized to 800x513 or 836x513 after entering the Wi-Fi menu.  I think it may depend on the width of the Wifi access point names, or length of message.

I'm attaching a video and logs from my device so that devs can further investigate why screen resizes incorrectly while on the Wi-Fi menu.

Notice in my video the "Back" button goes off screen to the right a bit while in the Wi-Fi menu.

https://user-images.githubusercontent.com/85074233/133454829-a6f855c3-b223-4b91-a1ed-2f3398601ca0.mov

[logfile.txt](https://github.com/Z-Bolt/OctoScreen/files/7170979/logfile.txt)

JeffB42, 2022-03-02:
I might be removing this screen, and making the WiFi information be set via ssh.  I'll look into this for 2.8 though.