Z-Bolt/OctoScreen #318: Print Time Does Not Update Every Second

AndyDiags, 2021-09-12:
Summary
The amount of time spend on the print only updates once every ~20 seconds instead of every second.

Steps
1. Begin a 5 hour print
2. View the amount of time spend on the print

Results
The amount of time spend on the print only updates once about every 20 seconds.  The time should update every send.  Screen is very responsive to touches as seen in attached video, so it does not seem that screen is locking up.

Note
Time left is inaccurate, but it does reflect the same time shown on the OctoPrint website, so I don't think it's an OctoScreen issue.  I would this ticket to focus on why the print time does not update every second.

OctoPrint v1.6.1
OctoScreen 2.7.4
OctoScreen plugin 2.6.0
Waveshare 4.3inch DSI LCD
Resoluton 800x480
Raspberry Pi 4 4GB
Prusa MK3S v3.10.0-4481

[Video.mp4](https://user-images.githubusercontent.com/85074233/132977736-bae336a3-5e56-476e-bbfa-4da551c84a88.mp4)
[logfile.zip](https://github.com/Z-Bolt/OctoScreen/files/7149228/logfile.zip)

AndyDiags, 2021-09-12:
Attaching follow up video, where I resolved the remaining time issue not matching, but the time elapsed is still not updating every second.

https://user-images.githubusercontent.com/85074233/132991109-80b4151e-b1a8-49af-a489-ddff7276cb63.mov

PS,
Installed M73 Override plug-in to resolve remaining time estimation on OctoPrint not matching Prusa LCD.

JeffB42, 2022-03-02:
I'm going to be working in the area for 2.8.  I'll try to get this issue resolved when I release it.