Z-Bolt/OctoScreen #319: OctoScreen occasionally reset during print

AndyDiags, 2021-09-18:
Summary:
OctoScreen occasionally reset during print, about ~once per hour.

Steps:
1. Start a print that takes a couple hours

Results:
About 50 minutes into the print, the screen restarted. The print continued as usual, and OctoScreen recovered.

Notes:
Taking a look at the full logs, I see this crash occurred a total of 4 times today.

```
Sep 18 17:47:02 octopi systemd[1]: octoscreen.service: Watchdog timeout (limit 40s)!
Sep 18 17:47:02 octopi systemd[1]: octoscreen.service: Killing process 5736 (xinit) with signal SIGABRT.
Sep 18 17:47:02 octopi xinit[5736]: (EE) Received signal 6 sent by process 1, uid 0
```
[syslog.zip](https://github.com/Z-Bolt/OctoScreen/files/7189656/syslog.zip)

OctoPrint v1.6.1
OctoScreen 2.7.4
OctoScreen plugin 2.6.0
Waveshare 4.3inch DSI LCD
Resoluton 800x480
Raspberry Pi 4 4GB
Prusa MK3S v3.10.0-4481

JeffB42, 2022-02-14:
I've been working on this, and this should be fixed in 2.7.5.  (sorry, no exact timeframe on the release just yet, but it's coming, and will keep this open until the release)

Codex-, 2022-03-19:
I get the same too, awesome to see there's a fix in progress. Looking forward to 2.8! 🎉

iammattcoleman, 2022-12-18:
I'm also affected by this.

@JeffB42, in case it helps: I tried increasing the watchdog timeout (I doubled it to 80s) and still experienced the issue.