Z-Bolt/OctoScreen #320: E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/x/xterm/xterm_344-1_armhf.deb   404   Not Found [IP: 93.93.128.193 80]

TMProjection, 2021-09-28:
Hi guys, ive tried this twice now

This is my second 3d printer that im installing octoscreen to but now im having trouble

when i try to run:
sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils

at the end i got:

E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/x/xterm/xterm_344-1_armhf.deb   404   Not Found [IP: 93.93.128.193 80]

and if i try that line again:
sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils

i will now get:

Reading package lists... Done
Building dependency tree
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libgtk-3-0 : Depends: adwaita-icon-theme but it is not going to be installed
              Depends: libatk-bridge2.0-0 (>= 2.5.3) but it is not going to be installed
              Depends: libatk1.0-0 (>= 2.15.1) but it is not going to be installed
              Depends: libcairo-gobject2 (>= 1.10.0) but it is not going to be installed
              Depends: libcolord2 (>= 0.1.10) but it is not going to be installed
              Depends: libcups2 (>= 1.6.0) but it is not going to be installed
              Depends: libepoxy0 (>= 1.4.3) but it is not going to be installed
              Depends: libjson-glib-1.0-0 (>= 0.12.0) but it is not going to be installed
              Depends: librest-0.7-0 (>= 0.7) but it is not going to be installed
              Depends: libsoup2.4-1 (>= 2.4.0) but it is not going to be installed
              Depends: libxcomposite1 (>= 1:0.3-1) but it is not going to be installed
              Depends: libgtk-3-common (>= 3.24.5-1+rpt2) but it is not going to be installed
              Recommends: libgtk-3-bin but it is not going to be installed
 octoscreen : Depends: libatk1.0-0 (>= 1.12.4) but it is not going to be installed
              Depends: libcairo-gobject2 (>= 1.10.0) but it is not going to be installed
 x11-xserver-utils : Depends: libxaw7 but it is not going to be installed
                     Depends: libxmu6 but it is not going to be installed
                     Depends: libxt6 but it is not going to be installed
 xserver-xorg : Depends: xserver-xorg-core (>= 2:1.17.2-2) but it is not going to be installed
                Depends: xserver-xorg-video-all but it is not going to be installed or
                         xorg-driver-video
                Depends: xserver-xorg-input-all but it is not going to be installed or
                         xorg-driver-input
                Depends: x11-xkb-utils but it is not going to be installed
                Recommends: xserver-xorg-legacy but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

i tried starting from scratch by reimaging the microsd card with same results, im using the same image file of octopi that i did when it worked around 9 months ago, so i dont see anything thats different besides the files thats being downloaded from the server using

sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils

any ideas?

Josef248, 2021-11-02:
Same Problem

Josef248, 2021-11-02:
In my case I followed the following steps:

`sudo apt update`
`sudo apt full-upgrade`
`sudo apt --fix-missing update`

And restart from zero the install instruction.
The installation completes successfully.

TMProjection, 2021-11-02:
> Did you ever figure this out? I'm experiencing something similar

nope, no one proposed a resolution either

i ended up cloning a working raspberry pi of mine with the same setup

JeffB42, 2022-02-06:
Sounds like you have it working then, so I'm going to close this.

closed 2022-02-06 by JeffB42

shaikhsafina, 2023-04-23:
facing same issue, tried almost all codes, still it is not fixed😥