Z-Bolt/OctoScreen #321: FR: Delete-Button in File-View, USB-Stick-Support

mr-who2012, 2021-09-28:
Hi there,
I want to add a feature-request: 

1) Is it possible to add a delete-button in the file-overview-screen to delete unwanted gcode-files? I am often trying to print the same part with slightly different properties, e.g. different infill, to sort out which one prints best. In such a case I don't want to turn on my computer, open the browser, connect to octoprints webpage and delete the unneccesary files. I just want to remove the fail-prints right-away.

2) Also: Support for USB-Sticks was nice, but I think this is an issue with octoprint itself. I installed usbmount to automatically mount hotplugged sticks into a subfolder of the upload-folder. This works fine. However, octoprint doesn't refresh the file-list regularly. So unplugging a stick doesn't get recognized.

3) Finally, when USB-Stick-Support was available: Move and Copy from and to stick was awesome.

I know, I actually made three suggestions, not one. But since SD-support is very limited in octoprint (very slow and with short filenames only), stick-support is a really good idea. Using sticks often is just more handy than webui-uploads.

JeffB42, 2022-02-06:
I'll add tis to the feature backlog, but you should know that file support might be limited, due to what can be done in OctoPrint, and what Marlin supports.

mr-who2012, 2022-04-27:
I don't know the technical details about this...but marlin shouldn't be involved at all. USB-stick support is only needed for the raspberry-pi-OS and octoprint.