Z-Bolt/OctoScreen #322: Adjust extrude speed buttons

cwgismo, 2021-10-05:
Where can I edit the speeds of the extrude buttons on the Touchscreen UI. In other words the UI has a button that toggles between  slow, medium and fast extrude speeds but I would like to edit the G code command so it extrudes slower. I find that even the slow setting in the UI is way to fast for my printer and causes jams. I have checked the Octoscreen and Octoprint config files and I don't see an option. Is there another text file I can edit. can someone point me in the right direction if this is possible.

AndyDiags, 2021-11-05:
Doing a quick search, I think you may be referring to this:
https://github.com/Z-Bolt/OctoScreen/blob/1cf8a705a9f0a4398562e244ea56c00ebf24fbdf/uiWidgets/FlowRateStepButton.go#L20-L22

Where slow is 75% the speed and fast 125% the speed.  Just modify it or add additional options if you'd like to go below 75%.

However if your printer currently jams at normal and slow sleeps, I would suggest you find the root cause of the jams rather than working around it by going extra slow.

cwgismo, 2023-04-25:
@AndyDiags My printer works fine during normal prints it is just when using the on screen (Octoprint) presets. I tried to find what you are referring to when I ssh into my Octoprint and I cannot find it. I tried looking in the /etc/octoscreen/config file and there is no settings for this. for example looking at the terminal in octoprint when I press the extrude button on octoscreen on the "slow" speed seting it sends a G1 E20 F300 command to the printer. basically I want to change the command to decrease the "F" (Federate) part of the command octoscreen sends (for example G1 E20 F150 instead of G1 E20 F300.

JeffB42, 2023-06-05:
@cwgismo if i remember correctly, OctoScreen uses a percentage (75%, 100%, 125%), and the rate used is what's defined in the printer's profile.  You need to go into OctoScreen, edit the printer's profile, then go to the "Axes" tab and set the rate there.

closed 2023-06-05 by JeffB42